thingsboard/thingsboard-gateway #458: [BUG]

abordbar, 2021-01-05:
**Describe the bug**

Hi Guys,
The bug I have found is as the following,
Support we have 3 modbus devices with address 1 , 2 , 3 on one bus and we have configure them in modbus-serial.json file with unitId 1,2,3.
if these 3 devices are online, the gateway work properly but if any of them be switched off, the gateway doesn't send the data of others.

**Connector name (If bug in the some connector):**
[e.g. OPC-UA Connector]

**Error traceback (If available):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

zbeacon, 2021-01-05:
Hi @abordbar ,

Thank you for your interest in ThingsBoard IoT gateway and message. I think it was caused by holding on connecting to unavailable device, I have added fix into master branch, please try to install the version of the gateway from it and rerun with a changed Modbus connector.

abordbar, 2021-01-05:
Thank you so much for your fast response and your efforts.

closed 2021-01-18 by zbeacon