thingsboard/thingsboard-gateway #460: write_registers() missing 1 required positional argument: 'values'"

anzaiok, 2021-01-12:
write_registers() missing 1 required positional argument: 'values'"
Traceback (most recent call last):
  File "F:\pycharmproject\thingsboard-gateway\thingsboard_gateway\connectors\modbus\modbus_connector.py", line 345, in __process_rpc_request
    response = self.__function_to_device(rpc_command_config, rpc_command_config["unitId"])
  File "F:\pycharmproject\thingsboard-gateway\thingsboard_gateway\connectors\modbus\modbus_connector.py", line 298, in __function_to_device
    result = self.__available_functions[function_code](address=config["address"],
TypeError: write_registers() missing 1 required positional argument: 'values'

when I use modbus connector to RPC method  I got the problem

zbeacon, 2021-01-18:
Hi @anzaiok ,

Thank you for your message, but please provide more information about your configuration and RPC method that you used.

arezola, 2021-02-08:
Dear @zbeacon ,

We have been using RPC calls in 16bit and 32bit variables (using function codes 6 and 16 respectively) with no problems before but we have just realized that RPC calls with functioncode 16 were not working properly. We have reviewed the error and it is directly related with this bug.

I imagine that it is due a Pymodbus library update, that asks for the positional argument "values" if the function code is 16 instead of the previous positional argument "value" that is the one used currently in line 304 of "modbud_connector.py" file.

We have just tested the following change in your code and it is working properly:

Change lines 307 to 311:

        elif function_code in (5, 6, 15, 16):
            print("value to values payload: "+str(config["payload"]))
            result = self.__available_functions[function_code](address=config["address"],
                                                               value=config["payload"],
                                                               unit=unit_id)

By this new lines (use value for function codes 5 and 6 and use values for function codes 15 y 16)

        elif function_code in (5, 6):
            result = self.__available_functions[function_code](address=config["address"],
                                                               value=config["payload"],
                                                               unit=unit_id)

        elif function_code in (15, 16):
            result = self.__available_functions[function_code](address=config["address"],
                                                               values=config["payload"],
                                                               unit=unit_id)

closed 2021-03-01 by zbeacon