thingsboard/thingsboard-gateway #463: [HELP] modbus function code 1 reads value "1" as "false"

Lautaro-a, 2021-01-19:
I want to read the state and control the relay of a device, which can be done trough the modbus address 0, a coil that becomes a 1 when the relay is on, and 0 when the relay is off. We've tested it with other modbus utilities and we can read it properly, but thingsboard always displays the value "false" when reading it, wether it is a 0 or a 1.
We also turned the relay on through thingsboard's RPC debug terminal, setting the value to 1, but in attributes the value still displays "false".

Connector name : modbus_serial.json
https://gist.githubusercontent.com/Lautaro-a/71ae2c81346217005cdce7fbf1aebc00/raw/60235165691d790cfaea2f91d14db150636b678f/modbus_serial.json

Versions:
 - OS: Debian GNU/linux 9.3
 - Thingsboard IoT Gateway version 2.5.2
 - Python version 3.5.3
 - Thingsboard version 3.1.1PE

chengdi123000, 2021-01-23:
I also encountered the same problem!

I used version 2.5.0 and add `"byteOrder":"LITTLE"` to every timeseries item. It works. 

I updated to the latest (version 2.5.5.2) using pip.  The coils are always `false`. So I have to downgrade it using

`pip install --upgrade thingsboard-gateway==2.5.0`

Lautaro-a, 2021-01-23:
Thank you @chengdi123000 ! it worked.
i added  "byteOrder":"LITTLE" to the "RelayState" attribute and now it's fixed, I'm still using Gateway version 2.5.2 so i didn't need to downgrade.
Here is the fixed connector in case anyone has the same problem: https://gist.githubusercontent.com/Lautaro-a/e5739436acd0f0011ff7eec47ceb2911/raw/5d88d19ebdcf2b2eac319804249716da3b8cd292/modbus_serial.json

chengdi123000, 2021-01-24:
I tried 2.5.5.2, it does not work for me. The latest version works for me is 2.5.4. There must be some commit breaks the connector.

closed 2021-08-04 by zbeacon

byte-rider, 2022-09-21:
I am also experiencing this issue. Not sure why it's been closed.