thingsboard/thingsboard-gateway #472: [HELP] failed to start ThingsBoard gateway

theone056, 2021-02-02:
#systemctl status thingsboard-gateway
things-gateway.service - ThingsBoard Gateway
Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: disabled)
Active: failed (Result: start-limit) since Mon 2021-02-01 13:11:22 PST; 19h ago
Main PID: 2088 (code=exited, status=1/FAILURE)

systemd[1]:  thingsboard-gateway.service: main process exited, code=exited, status=1/FAILURE
systemd[1]:  Unit thingsboard-gateway.service entered failed state.
systemd[1]:  thingsboard-gateway.service failed.
systemd[1]:  thingsboard-gateway.service holdoff time over, scheduling restart.
systemd[1]:  Stopped ThingsBoard Gateway.
systemd[1]:  start request repeated too quickly for thingsboard-gateway.service
systemd[1]   Failed to start ThingsBoard Gateway.
systemd[1]:  Unit thingsboard-gateway.service entered failed state.
systemd[1]:  thingsboard-gateway.service failed.

zbeacon, 2021-02-09:
Hi it looks like you have some issues in your configuration. You may check errors in the gateway logs. (/var/log/thingsboard-gateway/).

closed 2021-03-23 by zbeacon

schejbity, 2023-06-15:
Same issue.
What to do?

OS: Ubuntu 20.04

Used DEB instalation:
https://thingsboard.io/docs/iot-gateway/install/deb-installation/

`sudo systemctl status thingsboard-gateway.service`
```
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2023-06-15 13:06:55 CEST; 9min ago
    Process: 2197 ExecStart=/usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon() (code=exited, status=1/FAILURE)
   Main PID: 2197 (code=exited, status=1/FAILURE)

Jun 15 13:06:55 ubuntu systemd[1]: thingsboard-gateway.service: Scheduled restart job, restart counter is at 5.
Jun 15 13:06:55 ubuntu systemd[1]: Stopped ThingsBoard Gateway.
Jun 15 13:06:55 ubuntu systemd[1]: thingsboard-gateway.service: Start request repeated too quickly.
Jun 15 13:06:55 ubuntu systemd[1]: thingsboard-gateway.service: Failed with result 'exit-code'.
Jun 15 13:06:55 ubuntu systemd[1]: Failed to start ThingsBoard Gateway.
```

Logs are empty
```
-rw-rw-r-- 1 thingsboard_gateway thingsboard_gateway 0 Nov  2  2022 connector.log
-rw-rw-r-- 1 thingsboard_gateway thingsboard_gateway 0 Nov  2  2022 converter.log
-rw-rw-r-- 1 thingsboard_gateway thingsboard_gateway 0 Nov  2  2022 extension.log
-rw-rw-r-- 1 thingsboard_gateway thingsboard_gateway 0 Nov  2  2022 service.log
-rw-rw-r-- 1 thingsboard_gateway thingsboard_gateway 0 Nov  2  2022 storage.log
-rw-rw-r-- 1 thingsboard_gateway thingsboard_gateway 0 Nov  2  2022 tb_connection.log
```

When I try:
`cd /etc/thingsboard-gateway` and `sudo thingsboard-gateway`

It returns:
```
Traceback (most recent call last):
  File "/usr/bin/thingsboard-gateway", line 33, in <module>
    sys.exit(load_entry_point('thingsboard-gateway==3.2', 'console_scripts', 'thingsboard-gateway')())
  File "/usr/bin/thingsboard-gateway", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.8/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/__init__.py", line 15, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 50, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_connector import GrpcConnector
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/grpc_service/grpc_connector.py", line 18, in <module>
    from thingsboard_gateway.gateway.grpc_service.tb_grpc_manager import TBGRPCServerManager
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/grpc_service/tb_grpc_manager.py", line 24, in <module>
    from thingsboard_gateway.gateway.grpc_service.grpc_downlink_converter import GrpcDownlinkConverter
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/grpc_service/grpc_downlink_converter.py", line 21, in <module>
    from thingsboard_gateway.gateway.proto.messages_pb2 import *
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/proto/messages_pb2.py", line 33, in <module>
    _descriptor.EnumValueDescriptor(
  File "/usr/local/lib/python3.8/dist-packages/google/protobuf/descriptor.py", line 796, in __new__
[... 125 lines cut ...]
#  -
#    name: XMPP Connector
#    type: xmpp
#    configuration: xmpp.json
#
#  -
#   name: OCPP Connector
#   type: ocpp
#   configuration: ocpp.json
#
# ========= Customization ==========
#
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector
#
#  -
#    name: GRPC Connector 1
#    key: auto
#    type: grpc
#    configuration: grpc_connector_1.json
```