thingsboard/thingsboard-gateway #473: [HELP] My Modbus server only supports one thread.

gzlz, 2021-02-03:
**My Modbus server only supports one thread.**

My Modbus server only supports one thread.  For that only the first device telemetry is showed in thingsboard.
I would like to know if is possible to configure the modbus.json file in a way that force the gateway to do the devices requests one by one, only in one thread.

Thank you!

[modbus.json.txt](https://github.com/thingsboard/thingsboard-gateway/files/5919751/modbus.json.txt)

**Versions:**
-The gateway is running over the latest docker installation in a raspberry pi.
-The server is a Ubuntu Server 20.04.1 LTS]

Pawol85, 2021-03-03:
I have the same problem (BUG?)

samson0v, 2021-11-10:
Hi @gzlz, it seems from your config file that you are using the same unitId for different devices
[image]

Thanks for your interest in ThingsBoard IoT Gateway!

samson0v, 2021-12-02:
Hi @gzlz and @Pawol85, we remake the Modbus connector, so you can try and test it (using the version from the master branch)!
Pay attention, that there is a new config file for Modbus, to use it, you have to run Gateway with the old config file for Modbus and Gateway generate a new config file that will be named as "new_modbus.json" in the `config/` folder. And finally, in the `tb_gateway.yml` config file use the new Modbus config file.

If you have some questions, feel free to ask them!

samson0v, 2021-12-17:
Hi @gzlz and @Pawol85, did you try the new version of Modbus Connector, can we close the issue?

closed 2021-12-28 by zbeacon