thingsboard/thingsboard-gateway #474: modbus_connector - 301 With result Modbus Error

yvo-pfenninger, 2021-02-04:
Connected 3 modbus RTU devices. It worked fine until one device (device 2) was not reachable.

devices 1 and 3 stoped also and error 301 was reported:

 - DEBUG - [modbus_connector.py] - modbus_connector - 237 - Modbus trying connect to device_2
 - DEBUG - [modbus_connector.py] - modbus_connector - 301 - With result Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

{
        "name": "Modbus Default Server",
        "type": "serial",
        "method": "rtu",
        "port": "/dev/ttymxc0",
        "baudrate": 9600,
        "stopbits": 1,
        "bytesize": 8,
        "timeout": 35,
        "unitId": 102,
        "deviceName": "device_2",
        "deviceType": "type",
        "attributesPollPeriod": 3600000,
        "timeseriesPollPeriod": 60000,
        "sendDataOnlyOnChange": false,
        "attributes": [
            {.....

 - Docker on Debian 10 Buster
 - Thingsboard IoT Gateway version 2.5.5.2
 - Python version 3.7

zbeacon, 2021-02-09:
Hi @yvo-pfenninger ,

Could you try to use the version from the master branch? It should reconnect to the devices if one of them is unavailable.

yvo-pfenninger, 2021-02-09:
Hi @zbeacon 

I tried the "Installation from source" and had exactly the same error. It is not connecting unfortunately.

zbeacon, 2021-03-02:
Hi @yvo-pfenninger ,

I have checked one more time and in my environment, it processed the next device. It shows the error, but other devices are OK.

yvo-pfenninger, 2021-03-12:
Hi @zbeacon 

In my enviroment, it is not processing. What kind of setup do you use? Modbus RTU or TCP? Latest version tb gateway?

yvo-pfenninger, 2021-04-19:
Hi @zbeacon 
I installed once again the latest version from master branch. I tested it with RTU and TCP:
If I disconnect one of my devices, it is still not proceeding.