thingsboard/thingsboard-gateway #476: [BUG]

jfrancoVicomtech, 2021-02-08:
**Describe the bug**
The modbus connector doesn't work properly when sendDataOnlyOnChange is active and there are both timeseries and attributes in the configuration file for a device. The timeseries are never sent (the telemetry object is set to empty in to_send variable array when it is treated before the attributes).

I've solved this problem in code unindent a block of code in modbus_conector.py. I could do a pull request if i am given access to upload my branch.

Thanks and best regards

**Connector name (If bug in the some connector):**
[Modbus Connector]

**Versions (please complete the following information):**
 - Thingsboard IoT Gateway version 2.5.5
 - Python 3.7

zbeacon, 2021-02-09:
Hi @jfrancoVicomtech ,

Thank you for your investigation, please create a PR and I will merge it.

jfrancoVicomtech, 2021-02-10:
I've tried but it seems that I don't have permission to upload my branch

```
remote: Permission to thingsboard/thingsboard-gateway.git denied to jfrancoVicomtech.
fatal: unable to access 'https://github.com/thingsboard/thingsboard-gateway.git/': The requested URL returned error: 403
```

zbeacon, 2021-02-10:
Hi @jfrancoVicomtech ,

Yes, you cannot push changes directly to the repository, but you can create a fork, push changes to it and to your repository and then create pull request.

jfrancoVicomtech, 2021-02-12:
I made a the pull request #477 which seems to solve this bug

Thanks!

closed 2021-04-26 by zbeacon