thingsboard/thingsboard-gateway #485: WARNING - [tb_gateway_service.py] - Connector implementation not found for UDP Connector

corlinski, 2021-03-04:
**Describe the issue**
I've created a sample UDP connector and enabled it according to: 
https://thingsboard.io/docs/iot-gateway/custom/
When starting IoT GW I'm getting errors:
Mar 04 14:35:05 fliot-ns-prod-01 python3[8910]: ""2021-03-04 14:35:05" - WARNING - [tb_gateway_service.py] - tb_gateway_service - 312 - Connector implementation not found for UDP Connector"
Mar 04 14:35:05 fliot-ns-prod-01 python3[8910]: ""2021-03-04 14:35:05" - ERROR - [tb_utility.py] - tb_utility - 114 - Import CustomSerialConnector failed, path /usr/local/lib/python3.6/site-packages/thingsboard_gateway/connectors/serial doesn't exist"
Mar 04 14:35:05 fliot-ns-prod-01 python3[8910]: ""2021-03-04 14:35:05" - WARNING - [tb_gateway_service.py] - tb_gateway_service - 312 - Connector implementation not found for Custom Serial Connector"

I put connector implementation both in:
/usr/local/lib/python3.6/site-packages/thingsboard_gateway/connectors/
and 
/var/lib/thingsboard_gateway/extensions/

I've installed IoT GW according to docs:
https://thingsboard.io/docs/iot-gateway/install/rpm-installation/

Why is IoT GW looking for extension files in /usr/local/lib/python3.6/site-packages/thingsboard_gateway/connectors/ instead of 
/var/lib/thingsboard_gateway/extensions/ ?

**Configuration** (Attach your configuration file)
**tb_gateway.yaml**
  -
    name: UDP Connector
    type: udp
    configuration: udp.json
    class: UDPConnector
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#
  -
    name: Custom Serial Connector
    type: serial
    configuration: custom_serial.json
    class: CustomSerialConnector

**udp.json**
{
  "name": "UDP connector",
  "devices": [
    {
      "name": "UDP Tester",
      "type": "default",
      "converter": "UDPUplinkConverter",
...

**udp_connector.py in <udp> folder:**
"""Import libraries"""

import time
from threading import Thread
from random import choice
from string import ascii_lowercase
import socket

from thingsboard_gateway.connectors.connector import Connector, log    # Import base class for connector and logger
from thingsboard_gateway.tb_utility.tb_utility import TBUtility

class UDPConnector(Thread, Connector):    # Define a connector class, it should inherit from "Connector" class.
    def __init__(self, gateway, config, connector_type):
...

**Connector name (If you need help with some connector/converter):**
Custom UDP Connector
Custom Serial Connector

**Versions (please complete the following information):**
 - OS: CentOS Linux release 7.9.2009
 - Thingsboard IoT Gateway version 2.5.5.2
 - Python version 3.6.8

affoltep, 2021-03-14:
I have a similar problem with the development of a custom connector. The connector should read data from the gpsd using the gpsd library for python 3 [gpsd-py3](https://github.com/MartijnBraam/gpsd-py3)

The only warning message I get is:
""2021-03-14`` 18:55:13" - INFO - [tb_gateway_service.py] - tb_gateway_service - 77 - ThingsBoard IoT gateway version: 2.5.5.2"
""2021-03-14 18:55:13" - WARNING - [tb_gateway_service.py] - tb_gateway_service - 311 - Connector implementation not found for Custom Gnss Connector"
""2021-03-14 18:55:13" - INFO - [tb_gateway_service.py] - tb_gateway_service - 135 - Gateway `started."

I followed as well to the instructions for custom extensions. replaced the example "serial" by "gnss"
I had to copy the files folder gnss (including the adapted files custom_gnss_converter.py and custom_gnss_connector.py) to the folder /usr/lib/python3/dist-packages/thingsboard-gateway/extensions/ and /usr/lib/python3/dist-packages/thingsboard-gateway/connectors/ to get rid of some import error messages.

The service is starting with the messages above. However extensions.log, connector.log remain empty. 

So it seems to be the same issue as @corlinski have...thx for any help

corlinski, 2021-03-14:
hi. I think I managed to solve the problem by changing log level to DEBUG (somewhere in the main Config folder there's a file with logging settings).
it turned out there was a bug in the connector file i tried to use - one library was missing.

affoltep, 2021-03-14:
@corlinski thx, I will try that, too

closed 2021-03-15 by corlinski