thingsboard/thingsboard-gateway #486: Disconnection between MQTT Sever and Thinsboard Devices [HELP]

adriortega, 2021-03-05:
**Describe the issue**
Hi, I am having some problems with my mqtt connection. I am sending data from a remote installation to Thingsboard Gateway, using Modbus Protocol and a MQTT connection.

I am using several gateways simultaneously, each one connected to different devices, in order to read and send data; and also to send rpcs from Thingsboard  to the installation. There are 2 types of jsons:

- Some of them, have included only one devices with lots of different variables. These ones works properly in both issues, reading attributes and timeseries, and sending RPCs.
- The other ones, have a different configuration, because they include several devices, between 30 and 80 , but with much less variables each one. They send data to Thingsboard one after the other. I have noticed that if one device sends data 12-15 minutes ago, it disconnects from the server, so it is impossible to perform a rpc to it. 

So I would like to know if the mqtt server connection timeout is configurable, and where can I configure it. Because I want to run this application continuously, and it may be necessary to send rpcs at any moment.

Theoretically, when a rpc is made, a master's connection is created, so it shouldn´t be any problem.

**Configuration**

{
  "server": {
    "type": "tcp",
    "host": "10.101.1.33",
    "port": 502,
    "timeout": 1000,
    "method": "socket",
    "byteOrder": "BIG",
    "wordOrder": "LITTLE",
    "devices": [
      {
        "unitId": 1,
        "deviceName": "TCU_NCU_1_Baldio_1",
        "attributesPollPeriod": 86400000,
        "timeseriesPollPeriod": 300000,
        "sendDataOnlyOnChange": true,
        "attributes": [
          {
            "tag": "ActuatorCtrl",
            "type": "16uint",
            "functionCode": 4,
            "objectsCount": 1,
            "address": 30020,
            "sendDataOnlyOnChange": true
          }
        ],
        "timeseries": [
          {
            "tag": "EnergyToday_a1_J",
            "type": "32float",
            "functionCode": 4,
            "objectsCount": 2,
            "address": 30013,
            "sendDataOnlyOnChange": true
          }
   ],
        "rpc": {
          "setApproachBand_a1_pul": {
            "type": "16uint",
            "value": 0,
            "functionCode": 6,
            "address": 41096,
            "tag": "variable",
            "objectsCount": 1
          }
}

And this is repeated for all the devices in a json.

**Connector name (If you need help with some connector/converter):**

Our connector is modbus_connector.py file. 

**Error traceback (If it was raised):**
I have experienced two different failures or errors: 

Request URL: https://xxxxxxxxxxxxxxxx/api/plugins/rpc/twoway/23c36720-572f-11eb-9c4f-d5534ba62b01
Request Method: POST
Status Code: 408 
Remote Address: xx.xx.xxx.xxx:xxxx
Referrer Policy: strict-origin-when-cross-origin

Request URL: https://xxxxxxxxxxxxxxxx/api/plugins/rpc/twoway/23c36720-572f-11eb-9c4f-d5534ba62b01
Request Method: POST
Status Code: 409
Remote Address: xx.xx.xxx.xxx:xxxx
Referrer Policy: strict-origin-when-cross-origin

**Versions (please complete the following information):**
 - OS: Windows 10
 - Thingsboard IoT Gateway 2.5.5 version 
 - Python version: 3.7.8

samson0v, 2021-11-03:
Hi @adriortega, please, update your Gateway to the newest version via the master branch and let us know if it helps.
Thanks for your interest in ThingsBoard IoT Gateway.

samson0v, 2021-11-23:
Hi @adriortega, did it help, can we close the issue?

closed 2021-11-29 by zbeacon