thingsboard/thingsboard-gateway #493: [BUG] Custom modbus converter import fails

TheSamabo, 2021-03-17:
**Describe the bug**
I wrote a custom converter and i only extend the ModbusConverter class and try to get a log appear in logs.
Custom Converter: [https://pastebin.com/UXJYbjXK](https://pastebin.com/UXJYbjXK)
my modbus.json config: [https://pastebin.com/VynpNCt4](https://pastebin.com/VynpNCt4)

I think the problem is in how modbus_connector.py handles convertor import from config.
As someone new to this project the first example of custom converter is for MQTT converter that is listed on thingsboard.io documentation. That in any .json file you should specify an object with key value pairs for "type" ,"extension" maybe "extension-config"

**Connector name :**
Modbus Connector

**Error traceback:**
```
""2021-03-17 11:26:53" - ERROR - [modbus_connector.py] - modbus_connector - 106 - can only concatenate str (not "dict") to str"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 86, in __load_converters
    converter = TBUtility.check_and_import(self._connector_type, self.__config["converter"])(device)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_utility.py", line 77, in check_and_import
    if TBUtility.loaded_extensions.get(extension_type + module_name) is None:
TypeError: can only concatenate str (not "dict") to str
`
```

**Versions (please complete the following information):**
 - OS: PopOs 20.04
 - Thingsboard IoT Gateway version 2.5.5.2
 - Python version 3.8.5

TheSamabo, 2021-03-17:
Later i found modbus_converter.py takes the class name of the converter from the .json config. Like so: 
`"converter": "CustomConverterClassName",` and not not like this: `"converter": {"type": "custom", "extension": "CustomConverterClassName"}`. I still think this is kinda a bug. These things should be somehow unified into a more consistent format, i know it might not be possible for all connectors.

zbeacon, 2021-10-19:
Hi @TheSamabo ,

This issue relates to the previous version of the gateway, it should be fixed in the latest version, please try to use it or the master branch version. If the issue appears again - please open a new issue and link this issue to it.