thingsboard/thingsboard-gateway #494: ModbusTCP downlink bug [BUG]

jasonlee9900, 2021-03-19:
As for ModbusTCP, in case there are two devices in net, device A downlink works, device B broken, if try to control device B e.g. use control widgets, that will make whole net jam and collapse, and caused device A do not work until wait for long time e.g. several minutes to get response from device A, and success to send message to device A, although device A is good.

this happens for all versions of thingsboard 3.x + python 2.x.x

zbeacon, 2021-03-19:
Hi @jasonlee9900 ,

Thank you for your interest, we highly recommend to use the gateway with python 3.5 or above, could you please test your issue on the gateway with the python 3.5?

jasonlee9900, 2021-03-20:
my system working under python 3.9, I mean not python itself, I mean python gateway from 2.5.2 to 2.5.5 all has this bug: downlink will  jam and collapse everything.

jasonlee9900, 2021-03-20:
At very beginning of running gateway, if you only use gateway to operate good device, the gateway works good. But if you try to operate turn on/off through downlink to those broken devices (network cable plug out), the whole gateway will crazy to reconnecting and wait feedback or doing something else for long time, that make the whole gateway stop working and jam there, including stop communicate with those good device anymore. so you have to waiting for 5 or 10 minutes until gateway give up and go back to work with good device. sometimes, the gateway never go back (if you press on/off button many times) and just collapse! you have to stop the gateway and rerun gateway. 
As for remote control, people do not know whether a device is online or not, so, it's normal to press button to try whether it work or not, so I believe it's a bug.

samson0v, 2021-11-16:
Hi @jasonlee9900, please, update your Gateway via the master branch and let us know if it helps.
Thanks for your interest in ThingsBoard IoT Gateway.

samson0v, 2021-11-29:
Hi @jasonlee9900, did it help, can we close the issue?

jasonlee9900, 2021-11-30:
sorry, I have stopped testing this program long ago after one year working on it, and I may never come back with this program. sorry.

jasonlee9900, 2021-11-30:
Modbus Connector:

Added configuration parameters:

retries - Count of retries if polling was failed (default 3)

althougth I didn't test it, but I think your new update is wrong solution, modbus should never give up by 3 times polling try, it should always try for ever! Infinite times! But not that way your gateway did, spend so many resource on this polling try. the problem is not how many times polling try, the problem why it take some many resource on try! I do not know it is python limitation or something else, but I suggest you to limit polling try frequency, but try forever!

closed 2021-11-30 by zbeacon