thingsboard/thingsboard-gateway #495: [BUG] Problem with "bytes" type in Modbus

santiago-ruiz, 2021-03-19:
**Describe the bug**
I configured Modbus to read multiple registers with the `bytes` configuration but it's trying to decoded with `utf-8`, like the `string` type.

**Connector name:**
Modbus connector

**Configuration:**
```
{
  "server": {
    "byteOrder": "BIG",
    "devices": [
      {
        "attributes": [
          {
            "address": 1037,
            "functionCode": 3,
            "objectsCount": 1,
            "tag": "firmware_version",
            "type": "16uint"
          },
          {
            "address": 1140,
            "functionCode": 3,
            "objectsCount": 3,
            "tag": "mac",
            "type": "bytes"
          }
        ],
        "attributesPollPeriod": 3000,
        "deviceName": "abc124",
        "deviceType": "default",
        "sendDataOnlyOnChange": false,
        "timeseries": [
          {
           "address": 1100,
            "functionCode": 3,
            "objectsCount": 1,
            "tag": "test",
            "type": "16uint"
          }
        ],
        "timeseriesPollPeriod": 5000,
        "unitId": 1
      }
    ],
    "method": "rtu",
    "port": "/dev/ttyUSB0",
    "baudrate": 57600,
    "stopbits": 1,
    "bytesize": 8,
    "parity": "E",
    "strict": true,
    "timeout": 35,
    "type": "serial"
  }
}

```

**Error traceback:**
```
2021-03-19 16:08:13,948 - ERROR - [modbus_connector.py] - modbus_connector - 199 - 'utf-8' codec can't decode byte 0xdc in position 0: unexpected end of data
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 193, in __process_devices
    self.__gateway.send_to_storage(self.get_name(), to_send)
  File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 346, in send_to_storage
    json_data = dumps(data)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 382, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3/dist-packages/simplejson/encoder.py", line 296, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3/dist-packages/simplejson/encoder.py", line 378, in iterencode
    return _iterencode(o, 0)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xdc in position 0: unexpected end of data
```

**Versions:**
 - OS: Raspberry Pi OS 
 - Thingsboard IoT Gateway version 2.5.5.2 (last)
 - Python version 3.7

zbeacon, 2021-03-22:
Hi @santiago-ruiz ,

Thank you for your interest inThingsBoard IoT gateway.
Could you try to use the version from the master branch?

santiago-ruiz, 2021-03-22:
Hi @zbeacon, 
I've tried but with a similar result:
`UnicodeDecodeError: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
""2021-03-22 14:49:36" - ERROR - [modbus_connector.py] - modbus_connector - 199 - 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte"
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 193, in __process_devices
    self.__gateway.send_to_storage(self.get_name(), to_send)
  File "/usr/local/lib/python3.7/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 346, in send_to_storage
    json_data = dumps(data)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 382, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3/dist-packages/simplejson/encoder.py", line 296, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3/dist-packages/simplejson/encoder.py", line 378, in iterencode
    return _iterencode(o, 0)
`
I don't know why it's trying to decode `0xff`as a character like the `string` type does.

santiago-ruiz, 2021-03-23:
Hi @zbeacon,
I've upgraded to Thingsboard Gateway v2.6, but the problem is still there. On the other hand, I've tried to install version 2.5.1 from sources, where this feature was added but it didn't work either :( Maybe it's just a configuration problem that I'm not finding. Does this feature works for you??
Thank you

zbeacon, 2021-03-25:
Hi @santiago-ruiz ,

I cannot repeat this issue, it works after fix. It may be an issue with reinstalling. Please try to remove all versions of the gateway from your system and install it again. In order to remove you are able to use the following commands:

```
sudo pip3 uninstall thingsboard-gateway
pip3 uninstall thingsboard-gateway
```
Please run them until you will see the message like the following: `thingsboard-gateway not found` (On root level and on user level). 
Then you can install the gateway using pip or any option that you like and run it.
After start please make sure that you have the gateway version 2.6 (You can check it in service.log file at starting gateway).

santiago-ruiz, 2021-03-29:
Hi @zbeacon,
After a fresh installation of OS and thingsboard-gateway everything is working smoothly. Thank you!

closed 2021-03-29 by santiago-ruiz