thingsboard/thingsboard-gateway #498: [BUG] Modbus RPC still can't write value 0 on coil after version 2.6

Lautaro-a, 2021-04-06:
**Describe the bug**
Modbus RPC allows to write value "1" on coil but writing "0" is sent as "null" and returns error. Supposedly this was fixed in release 2.6 as stated here https://github.com/thingsboard/thingsboard-gateway/commit/fd24937f1844cc4b7d86f430695e155fa83e7bda , but after update it's still not working.

[image]
[image]

Connector: modbus_serial.json
https://gist.githubusercontent.com/Lautaro-a/71ae2c81346217005cdce7fbf1aebc00/raw/60235165691d790cfaea2f91d14db150636b678f/modbus_serial.json

Error traceback :
```
""2021-04-06 13:27:51" - ERROR - [modbus_connector.py] - modbus_connector - 322 - 'value'"
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 310, in server_side_rpc_handler
    self.__process_rpc_request(content, rpc_command_config)
  File "/usr/local/lib/python3.5/dist-packages/thingsboard_gateway/connectors/modbus/modbus_connector.py", line 332, in __process_rpc_request
    rpc_command_config, content)
  File "/usr/local/lib/python3.5/dist-packages/thingsboard_gateway/connectors/modbus/bytes_modbus_downlink_converter.py", line 51, in convert
    value = config["value"]
KeyError: 'value'
```

Versions:

OS: Debian GNU/linux 9.3
Thingsboard IoT Gateway version 2.6
Python version 3.5.3
Thingsboard version 3.1.1PE

zbeacon, 2021-04-14:
Hi @Lautaro-a ,

Please try the version from the master branch.

zbeacon, 2021-04-14:
Or just as a workaround - you can set parameter "value" for RPC reuqest configuration in modbus configuration json file to 0.

closed 2021-09-06 by zbeacon