thingsboard/thingsboard-gateway #499: [BUG] unstable connectivity with AWS IOT via mqtt connector

t0mmyz7, 2021-04-14:
Hi
I am using MQTT connector with AWS IOT

This was working fine for me but since latest version of tb-gw, my connection keeps connecting/closing.

this is my config:
```
{
  "broker": {
    "host": "xxxxx-ats.iot.eu-central-1.amazonaws.com",
    "name": "Default Local Broker",
    "port": 8883,
    "security": {
      "caCert": "/etc/thingsboard-gateway/ca.pem",
      "cert": "/etc/thingsboard-gateway/cert.pem",
      "privateKey": "/etc/thingsboard-gateway/key.pem"
    },
    "ssl": true
  },

```

logs:
```

Apr 14 09:23:46 ip-172-31-2-103 python3[9976]: ""2021-04-14 09:23:46" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 125 - Disconnected client: <paho.mqtt.client.Client objec
Apr 14 09:23:47 ip-172-31-2-103 python3[9976]: ""2021-04-14 09:23:47" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
Apr 14 09:23:47 ip-172-31-2-103 python3[9976]: ""2021-04-14 09:23:47" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 125 - Disconnected client: <paho.mqtt.client.Client objec
Apr 14 09:23:48 ip-172-31-2-103 python3[9976]: ""2021-04-14 09:23:48" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
Apr 14 09:23:49 ip-172-31-2-103 python3[9976]: ""2021-04-14 09:23:49" - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 125 - Disconnected client: <paho.mqtt.client.Client objec
Apr 14 09:23:50 ip-172-31-2-103 python3[9976]: ""2021-04-14 09:23:50" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
```

```
"2021-04-14 09:20:13" - DEBUG - [mqtt_connector.py] - mqtt_connector - 254 - Converter JsonMqttUplinkConverter for topic /thingsboard/ingics/ibs03 - found!"
""2021-04-14 09:20:13" - INFO - [mqtt_connector.py] - mqtt_connector - 274 - Connector "MQTT Broker Connector" subscribe to /thingsboard/ingics/ibs03"
""2021-04-14 09:20:13" - INFO - [mqtt_connector.py] - mqtt_connector - 322 - "MQTT Broker Connector" subscription success to topic /thingsboard/ingics/ibs03tp, subscription message id = 1"
""2021-04-14 09:20:13" - INFO - [mqtt_connector.py] - mqtt_connector - 322 - "MQTT Broker Connector" subscription success to topic /thingsboard/ingics/ibs03, subscription message id = 2"
""2021-04-14 09:20:19" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03"
""2021-04-14 09:20:22" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03"
""2021-04-14 09:20:22" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03"
""2021-04-14 09:20:33" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03"
""2021-04-14 09:20:39" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03"
- [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03"
""2021-04-14 09:22:19" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03"
""2021-04-14 09:22:22" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03"
""2021-04-14 09:22:22" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03tp"
""2021-04-14 09:22:22" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03"
""2021-04-14 09:22:28" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03tp"
""2021-04-14 09:22:28" - DEBUG - [mqtt_connector.py] - mqtt_connector - 353 - Successfully converted message from topic /thingsboard/ingics/ibs03tp"
""2021-04-14 09:22:28" - DEBUG - [mqtt_connector.py] - mqtt_connector - 308 - "MQTT Broker Connector" was disconnected. (<paho.mqtt.client.Client object at 0x7fd31caec7b8>, None, 0)"
""2021-04-14 09:22:28" - INFO - [mqtt_connector.py] - mqtt_connector - 206 - MQTT Broker Connector has been stopped."

```

any idea?

t0mmyz7, 2021-04-14:
or am I confusing here the tb_mqtt connection and the mqtt connector, and what I am seeing is the connect/disconnect of the gw to tb?

zbeacon, 2021-04-14:
HI @t0mmyz7 ,

It looks like the gateway client disconnects from ThingsBoard because of the previous session wasn't finished correctly, so it tries to reconnect and receives disconnect event again. This bug doesn't relates to AWS, please try to use integration on PE version.

t0mmyz7, 2021-04-14:
Can we try to debug the connection to TB issue somehow?

t0mmyz7, 2021-04-18:
@zbeacon this was working fine with previous version

any idea why it has unstable connection with TB?