thingsboard/thingsboard-gateway #500: thingsboard-gateway mqtt problem[HELP]

zhangxujun16, 2021-04-15:
**Encount a problem with thingsboard-gateway mqtt**
   The  tb_gateway.yaml and the default mqtt.josn like following:
**Configuration**
thingsboard:
  host: 192.168.0.72
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  security:
    accessToken: InVxxxxxxxxxxxxxx
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000

connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json
....

{
  "broker": {
    "name":"Default Local Broker",
    "host":"192.168.0.72",
    "port":1884,
    "clientId": "ThingsBoard_gateway",
    "security": {
      "type": "basic",
      "username": "admin",
      "password": "public"
    }
  },
  "mapping": [
    {
      "topicFilter": "/sensor/data",
      "converter": {
        "type": "json",
        "deviceNameJsonExpression": "${serialNumber}",
        "deviceTypeJsonExpression": "${sensorType}",
        "timeout": 60000,
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          },
          {
            "type": "string",
            "key": "${sensorModel}",
            "value": "on"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    },
....

I publish topic and payload likes:
topic:   /sensor/data
payload:  {“serialNumber”: “SN-001”, “sensorType”: “Thermometer”, “sensorModel”: “T1000”, “temp”: 42, “hum”: 58}

**Get an error :   "deviceName is empty" **
root@ubuntu:/var/log/thingsboard-gateway# systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2021-04-15 06:37:12 UTC; 6min ago
   Main PID: 2953 (python3)
      Tasks: 8 (limit: 973)
     CGroup: /system.slice/thingsboard-gateway.service
             └─2953 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

Apr 15 06:37:43 ubuntu python3[2953]:     body = loads(body)
Apr 15 06:37:43 ubuntu python3[2953]:   File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
Apr 15 06:37:43 ubuntu python3[2953]:     return _default_decoder.decode(s)
Apr 15 06:37:43 ubuntu python3[2953]:   File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
Apr 15 06:37:43 ubuntu python3[2953]:     obj, end = self.raw_decode(s)
Apr 15 06:37:43 ubuntu python3[2953]:   File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
Apr 15 06:37:43 ubuntu python3[2953]:     return self.scan_once(s, idx=_w(s, idx).end())
Apr 15 06:37:43 ubuntu python3[2953]: simplejson.errors.JSONDecodeError: Expecting property name enclosed in double quotes: line 1 column 2 (char 1)
Apr 15 06:37:43 ubuntu python3[2953]: ""2021-04-15 06:37:43" - ERROR - [tb_utility.py] - tb_utility - 61 - **deviceName is empty** in data: {"deviceName": null, "deviceType": null, "attributes": [], "telemetry": []}"
Apr 15 06:37:43 ubuntu python3[2953]: ""2021-04-15 06:37:43" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 322 - Data from MQTT Broker Connector connector is invalid."

**Converter.log**

""2021-04-15 06:37:43" - ERROR - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 77 - Error in converter, for config: 
{"type": "json", "deviceNameJsonExpression": "${serialNumber}", "deviceTypeJsonExpression": "${sensorType}", "timeout": 60000, "attributes": [{"type": "string", "key": "model", "value": "${sensorModel}"}, {"type": "string", "key": "${sensorModel}", "value": "on"}], "timeseries": [{"type": "double", "key": "temperature", "value": "${temp}"}, {"type": "double", "key": "humidity", "value": "${hum}"}]}
 and message: 
{“serialNumber”: “SN-001”, “sensorType”: “Thermometer”, “sensorModel”: “T1000”, “temp”: 42, “hum”: 58}
"
""2021-04-15 06:37:43" - ERROR - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 78 - Expecting property name enclosed in double quotes: line 1 column 2 (char 1)"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py", line 62, in convert
    value = TBUtility.get_value(datatype_config["value"], data, datatype_config["type"], expression_instead_none=True)
  File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_utility/tb_utility.py", line 129, in get_value
    body = loads(body)
  File "/usr/lib/python3/dist-packages/simplejson/__init__.py", line 518, in loads
    return _default_decoder.decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 370, in decode
    obj, end = self.raw_decode(s)
  File "/usr/lib/python3/dist-packages/simplejson/decoder.py", line 400, in raw_decode
    return self.scan_once(s, idx=_w(s, idx).end())
simplejson.errors.JSONDecodeError: Expecting property name enclosed in double quotes: line 1 column 2 (char 1)

**Versions :**
 - OS: [ Ubuntu 18.04] 
Linux ubuntu 5.4.0-1033-raspi #36-Ubuntu SMP PREEMPT Thu Mar 25 19:00:51 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
q          (Raspberry pie 3B+)

 - Thingsboard IoT Gateway version [2.6]
 - Python version[ 3.8]

Thanks!

zbeacon, 2021-04-15:
Hi @zhangxujun16 ,

The gateway cannot parse your message, because you use wrong quotes:
Expecting property name enclosed in double quotes: line 1 column 2 (char 1)"

Please use " instead of “ and ” - these are different symbols.

zhangxujun16, 2021-04-16:
Thank you very much!  It's OK now!

closed 2021-04-28 by zbeacon