thingsboard/thingsboard-gateway #503: Thingsboard-gateway opcua [HELP]

zhangxujun16, 2021-04-19:
**Not recieve remote data from opcua server**
  
  I have run a opcua server to simulate a remote deive. 
**opcua server code in python :**
```
import sys

sys.path.insert(0, "..")
import time
from opcua import ua, Server, uamethod

@uamethod
def set_value(parent, node_id, value):
    node = server.get_node(node_id)
    res = False
    if ua.AccessLevel.CurrentWrite in list(node.get_access_level()):
        node.set_value(value)
        res = True
    return res

def argument(name, date_type, value_rank):
    v = ua.Argument()
    v.Name = name
    v.DataType = ua.NodeId(date_type)
    v.ValueRank = value_rank
    v.ArrayDimensions = []
    v.Description = ua.LocalizedText(name)
    return v

if __name__ == "__main__":

    global server
    server = Server()

    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    server.set_security_policy([
        ua.SecurityPolicyType.NoSecurity,
        ua.SecurityPolicyType.Basic128Rsa15_SignAndEncrypt,
        ua.SecurityPolicyType.Basic128Rsa15_Sign])
    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)
    objects = server.get_objects_node()
     dev = objects.add_object(idx, "Device")
    dev_name = dev.add_variable(idx, "Name", "opcua_sensor")
    dev_battery = dev.add_variable(idx, "BatteryLevel", 9.0)
    dev_humidity = dev.add_variable(idx, "Humidity", 13)
    dev_switch = dev.add_variable(idx, "Switch", 1)
    dev_temperature = dev.add_variable(idx, "Temperature", 19)

    dev_switch.set_writable()

    in_arg_node_id = argument("node_id", ua.ObjectIds.String, -1)
    in_arg_value = argument("value", ua.ObjectIds.Int64, -1)
    out_arg = argument("result", ua.ObjectIds.Boolean, -1)

    set_value_node = dev.add_method(idx, "set_value", set_value, [in_arg_node_id, in_arg_value], [out_arg])

    server.start()

    try:
        count = 0
        while True:
            time.sleep(1)
            count += 0.1
            dev_battery.set_value(count)
    finally:
        server.stop()

```
I check opecua server with UaExpert , and it works well .

[image]

tb_gateway.yaml
thingsboard:
host: 192.168.0.57
port: 1883
remoteShell: false
remoteConfiguration: false
security:
accessToken: InVDsr4E5APwXN9XoEhx
qos: 1
storage:
type: memory
read_records_count: 100
max_records_count: 100000
connectors:
-
    name: OPC-UA Connector
    type: opcua
    configuration: opcua.json 
 
**opcua.json**
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "localhost:4840/freeopcua/server/",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions":false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Device1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "attributes": [
          {
            "key": "temperature °C",
            "path": "${ns=2;i=5}"
          }
        ],
        "timeseries": [
          {
            "key": "humidity",
            "path": "${Root\\.Objects\\.Device1\\.TemperatureAndHumiditySensor\\.Humidity}"
          },
          {
            "key": "batteryLevel",
            "path": "${Battery\\.batteryLevel}"
          }
        ],
        "rpc_methods": [
          {
            "method": "multiply",
            "arguments": [2, 4]
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.Device1\\.serialNumber"
          }
        ]
      }
    ]
  }
}
```
**logs  (/var/log/thingsboard-gateway/)**
[image]

connector.log
""2021-04-19 05:53:26" - INFO - [opcua_connector.py] - opcua_connector - 96 - Starting OPC-UA Connector"
""2021-04-19 05:53:26" - INFO - [opcua_connector.py] - opcua_connector - 121 - OPC-UA connector OPC-UA Connector connected to server localhost:4840/freeopcua/server/"

service.log
""2021-04-19 05:53:25" - INFO - [tb_gateway_service.py] - tb_gateway_service - 72 - Gateway starting..."
""2021-04-19 05:53:25" - INFO - [tb_gateway_service.py] - tb_gateway_service - 77 - ThingsBoard IoT gateway version: 2.6"
""2021-04-19 05:53:26" - INFO - [tb_utility.py] - tb_utility - 111 - Import OpcUaConnector from /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/opcua."
""2021-04-19 05:53:26" - INFO - [tb_gateway_service.py] - tb_gateway_service - 135 - Gateway started."

tb_connection.log
""2021-04-19 05:53:25" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 1"
""2021-04-19 05:53:25" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"

** thingsboard UI **
[image]

[image]

**Versions:**
 - OS: [Ubuntu 18.04]
    Linux ubuntu 5.4.0-1034-raspi #37-Ubuntu SMP PREEMPT Mon Apr 12 23:14:49 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

 - Thingsboard IoT Gateway version [2.6]
 - Python version[3.8]

zhangxujun16, 2021-04-20:
The tb_gateway2.6 does not support this opcua.json format.  
[image]
The path should be replaced by ns and i.

closed 2021-04-20 by zhangxujun16

zbeacon, 2021-04-26:
Hi @zhangxujun16 ,

Thank you for your interest in ThingsBoard IoT gateway. Could you try to use ```"${Root\\.Objects\\.Device1}"``` instead of ``` "Root\\.Objects\\.Device1"``` for **deviceNodePattern** parameter and try again?