thingsboard/thingsboard-gateway #505: Problem to cutomize the gateway [HELP]

mrroot998, 2021-04-22:
Hello everybody,

I'm encoutering a problem in customizing the Thingsboard IoT Gateway. I Installed on my mosquitto mqtt Broker a [message-timestamp](https://github.com/eclipse/mosquitto/tree/master/plugins/message-timestamp) plugin to add a properties field on each messages send on the broker (but need to use MQTTv5). I also modified this plugin to add a client_id properties instead of timestamp properties.

The problem is when I modify the code of the mqtt connector of the gateway ([mqtt_connector.py](https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/connectors/mqtt/mqtt_connector.py)) to be able to get the properties field (which is add by the plugin). 
I have to specify the protocol MQTTv5 in the initialization of the mqtt client object and the adding of this parameter make the on_message callback untriggered.
Why does it act like this and how can I to correct this ?

I did a first code in python to be sure that the plugin works and to know how to be able to receive this properties field:

```
import paho.mqtt.subscribe as subscribe
import paho.mqtt.client as Client
import paho.mqtt.properties as properties
import struct
import json

def on_message(client, userdata, message):
	print("Topic: %s \nPayload: %s \nMessage properties: %s" % (message.topic, message.payload, message.properties))

host = "xxx.xxx.xxx.xxx"
port = 1883
topic = "/sensor/1/data"

client = Client.Client("subs_client", protocol=Client.MQTTv5)
client.on_message = on_message

resultConnect = client.connect(host,port,60)
if resultConnect == 0:
	print("connected to: " + host + " on port: " + str(port))
else:
	print("error occurred while connecting, code error: " + str(resultConnect))

resultSubscribe = client.subscribe(topic)
if resultSubscribe[0] == 0:
	print("successfully subscribed to: " + topic)
try:
	client.loop_forever()
except KeyboardInterrupt:
	print("\nProgram interrupted by Keyboard !")
	exit(0)
```

and when I publish a simple message with mosquitto:

`mosquitto_pub -h xxx.xxx.xxx.xxx -p 1883 -i pub_client -t /sensor/1/data -m '{"temp": 14.2, "hum": 7}'`

the result is:

`connected to: xxx.xxx.xxx.xxx on port: 1883`
`successfully subscribed to: /sensor/1/data`
`Topic: /sensor/1/data `
`Payload: b'{"temp": 14.2, "hum": 7}' `
`Message properties: [UserProperty : [('client_id', 'pub_client')]]`

So the code is correct and works well.

And when I specify the version of the mqtt protocol in the connector code, the connector log only display that:

```
''2021-04-21 14:39:03' - INFO - mqtt_connector - 217 - MQTT Broker Connector has been stopped.'
''2021-04-21 14:39:03' - DEBUG - tb_logger - 37 - Added remote handler to log extension'
''2021-04-21 14:39:03' - DEBUG - tb_logger - 37 - Added remote handler to log converter'
''2021-04-21 14:39:03' - DEBUG - tb_logger - 37 - Added remote handler to log connector'
```

(after a service restart and the publication of a message on the "/sensor/+/data" default topic of the gateway)

and the service log display that:

[... 31 lines cut ...]
''2021-04-21 14:39:03' - DEBUG - tb_utility - 100 - ModuleSpec(name='CustomMqttUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f4a1807dc18>, origin='/usr/local/lib/python3.6/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py')'
''2021-04-21 14:39:03' - DEBUG - tb_utility - 107 - <module 'CustomMqttUplinkConverter' from '/usr/local/lib/python3.6/dist-packages/thingsboard_gateway/connectors/mqtt/mqtt_connector.py'>'
''2021-04-21 14:39:03' - DEBUG - tb_utility - 100 - ModuleSpec(name='CustomMqttUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f4a1814a470>, origin='/usr/local/lib/python3.6/dist-packages/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py')'
''2021-04-21 14:39:03' - DEBUG - tb_utility - 107 - <module 'CustomMqttUplinkConverter' from '/usr/local/lib/python3.6/dist-packages/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py'>'
''2021-04-21 14:39:03' - DEBUG - tb_utility - 100 - ModuleSpec(name='CustomMqttUplinkConverter', loader=<_frozen_importlib_external.SourceFileLoader object at 0x7f4a1814a748>, origin='/usr/local/lib/python3.6/dist-packages/thingsboard_gateway/extensions/mqtt/custom_mqtt_uplink_converter.py')'
''2021-04-21 14:39:03' - DEBUG - tb_utility - 107 - <module 'CustomMqttUplinkConverter' from '/usr/local/lib/python3.6/dist-packages/thingsboard_gateway/extensions/mqtt/custom_mqtt_uplink_converter.py'>'
''2021-04-21 14:39:03' - INFO - tb_utility - 111 - Import CustomMqttUplinkConverter from /usr/local/lib/python3.6/dist-packages/thingsboard_gateway/extensions/mqtt.'
''2021-04-21 14:39:03' - DEBUG - tb_gateway_service - 233 - Received data: {'client': {'firmware_version': '1.0.1'}, 'shared': {'device': '1', 'hum': 40, 'RemoteLoggingLevel': 'DEBUG', 'data': {'temp': 82.89, 'hum': 90}}}'
''2021-04-21 14:39:03' - INFO - tb_gateway_service - 256 - Remote logging has being updated. Current logging level is: DEBUG '
''2021-04-21 14:39:03' - DEBUG - tb_gateway_service - 258 - Shared attributes received (device, hum, RemoteLoggingLevel, data).'
''2021-04-21 14:39:03' - DEBUG - tb_gateway_service - 260 - Client attributes received (firmware_version).'
''2021-04-21 15:39:04' - DEBUG - tb_updater - 69 - Checking for new version'

```

(Attached to this post, the file of the connector.py with my modification (the line where the problem occures is on the 92nd one).
[mqtt_connector.txt](https://github.com/thingsboard/thingsboard-gateway/files/6357493/mqtt_connector.txt)

In thanking you for your help.

**Versions:**
 - OS: [Ubuntu 18.04]
 - Thingsboard IoT Gateway version [2.6]
 - Paho MQTT version[1.5.1]
 - Python version[3.6]

zbeacon, 2021-04-26:
Hi @mrroot998 ,

Thank you for your interest in ThingsBoard IoT gateway.

Please try to use the version from the master branch, or just add into on_subscribe function one more argument (it appears only on version MQTTv5).

mrroot998, 2021-04-26:
Hi @zbeacon,

Thanks for your answer.

I just tried to run thingsboard-gateway (from the git repository) with the good settings and when I run the python script tb_gateway.py I have the following error:

```
""2021-04-26 15:54:23" - INFO - [mqtt_connector.py] - mqtt_connector - 241 - MQTT Broker Connector connected to mqtt.test.perax.com:1883 - successfully."
""2021-04-26 15:54:23" - INFO - [tb_utility.py] - tb_utility - 111 - Import JsonMqttUplinkConverter from /usr/local/lib/python3.6/dist-packages/thingsboard_gateway/connectors/mqtt."
""2021-04-26 15:54:23" - INFO - [mqtt_connector.py] - mqtt_connector - 282 - Connector "MQTT Broker Connector" subscribe to /sensor/+/data"
""2021-04-26 15:54:23" - INFO - [mqtt_connector.py] - mqtt_connector - 282 - Connector "MQTT Broker Connector" subscribe to /sensor/+/type/+/data"
""2021-04-26 15:54:23" - INFO - [mqtt_connector.py] - mqtt_connector - 282 - Connector "MQTT Broker Connector" subscribe to /sensor/data"
""2021-04-26 15:54:23" - INFO - [mqtt_connector.py] - mqtt_connector - 282 - Connector "MQTT Broker Connector" subscribe to /sensor/+/data"
""2021-04-26 15:54:23" - INFO - [tb_utility.py] - tb_utility - 111 - Import CustomMqttUplinkConverter from /usr/local/lib/python3.6/dist-packages/thingsboard_gateway/extensions/mqtt."
""2021-04-26 15:54:23" - INFO - [mqtt_connector.py] - mqtt_connector - 282 - Connector "MQTT Broker Connector" subscribe to /custom/sensors/+"
Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.6/dist-packages/paho/mqtt/client.py", line 3452, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python3.6/dist-packages/paho/mqtt/client.py", line 1779, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/local/lib/python3.6/dist-packages/paho/mqtt/client.py", line 1181, in loop
    rc = self.loop_read(max_packets)
  File "/usr/local/lib/python3.6/dist-packages/paho/mqtt/client.py", line 1572, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.6/dist-packages/paho/mqtt/client.py", line 2310, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.6/dist-packages/paho/mqtt/client.py", line 2944, in _packet_handle
    return self._handle_suback()
  File "/usr/local/lib/python3.6/dist-packages/paho/mqtt/client.py", line 3148, in _handle_suback
    self, self._userdata, mid, reasoncodes, properties)
TypeError: _on_subscribe() takes 5 positional arguments but 6 were given

""2021-04-26 15:54:23" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
""2021-04-26 15:54:23" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 2"
""2021-04-26 15:54:23" - INFO - [tb_gateway_service.py] - tb_gateway_service - 256 - Remote logging has being updated. Current logging level is: DEBUG "
^C""2021-04-26 16:50:00" - INFO - [tb_gateway_service.py] - tb_gateway_service - 214 - Stopping..."
""2021-04-26 16:50:00" - INFO - [mqtt_connector.py] - mqtt_connector - 214 - MQTT Broker Connector has been stopped."
""2021-04-26 16:50:00" - INFO - [tb_gateway_service.py] - tb_gateway_service - 216 - The gateway has been stopped."

```

Here is the configuration:

* for mqtt.json:
   ```
   "broker": {
      "name":"MosquittoBroker",
      "host":"xxxxxxxxxx",
      "port":1883,
      "clientId": "github-ThingsBoard_gateway_",
      "security": {
        "type": "basic",
        "username": "xxxxxxxx",
        "password": "xxxxxxxxx"
      }
     ```
* for tb_gateway.yaml:
   ```
      thingsboard:
        host: xxxxxxxxxxx
        port: 1883
        remoteShell: false
        remoteConfiguration: false
        statsSendPeriodInSeconds: 3600
        security:
          accessToken: xxxxxxxxxxxxx
        qos: 1
      storage:
        type: memory
        read_records_count: 100
        max_records_count: 100000
      #  type: file
      #  data_folder_path: ./data/
      #  max_file_count: 10
      #  max_read_records_count: 10
      #  max_records_per_file: 10000
      connectors:
        -
          name: MQTT Broker Connector
          type: mqtt
          configuration: mqtt.json
   ```

mrroot998, 2021-05-06:
I couldn't correct the error with thingsboard (from github directory) but in adding the new argument in the on_subscribe() function for the ubuntu package version, I could correct my principal error.