thingsboard/thingsboard-gateway #506: [BUG] Gateway pushing data to cloud too often

santiago-ruiz, 2021-04-23:
**Describe the bug**
I'm currently using ThingsBoard Cloud with a ThingsBoard Gateway and looking at the API usage I found that the "Transport hourly activity" was higher than expected. After searching for a bit, looking at the MQTT traffic in the gateway with `tcpdump` I found that is pushing datapoints (`eventsProduced`, `eventsSent`, `modbusconnectorEventsProduced`, and `modbusconnectorEventsSent`) every 5 seconds. This frequency cannot be changed in configuration and represents more than 2M Data Points in transport monthly. This limits the API usage a user can get from ThingsBoard Cloud and its unnecessary information. I think this should be configurable in some way....

**Versions :**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version  2.5.5.2
 - Python version 3.7

zbeacon, 2021-04-26:
Hi @santiago-ruiz ,

Thank you for your interest in ThingsBoard IoT gateway. I have added the functionality like you described, to use it please try to use the version from the master branch. Also it will be available in the next release.

santiago-ruiz, 2021-04-26:
Hi @zbeacon, thanks for the quick answer. This new feature is working perfectly!

closed 2021-04-26 by santiago-ruiz