thingsboard/thingsboard-gateway #508: [HELP]

jonathanwvd, 2021-04-28:
Hello,

I have been trying to connect and receive data from an OPC-UA server. First, I tried to use Docker on Linux, but I got the following error:
Connection refused on connection to OPC-UA server with url opc.tcp://127.0.0.1:4840/freeopcua/server/"

Then, I tried to use the Python version. I was able to connect to the OPC server, but I am not getting any data.

I am using the default OPC-UA server example from opcua library as follows:
```
import sys

sys.path.insert(0, "..")
import time

from opcua import ua, Server

if __name__ == "__main__":

    # setup our server
    server = Server()
    server.set_endpoint("opc.tcp://localhost:4840/freeopcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    idx = server.register_namespace(uri)
    device = server.nodes.objects.add_object(idx, "Device1")
    name = server.nodes.objects.add_variable(idx, "serialNumber", "TEST")
    name.set_writable()
    temperature_and_humidity = device.add_object(idx, "TemperatureAndHumiditySensor")
    temperature = temperature_and_humidity.add_variable(idx, "Temperature", 56.7)
    humidity = temperature_and_humidity.add_variable(idx, "Humidity", 68.7)
    battery = device.add_object(idx, "Battery")
    battery_level = battery.add_variable(idx, "batteryLevel", 24)

    # starting!
    server.start()

    try:
        count = 0
        while True:
            time.sleep(1)
            count += 0.1
            temperature.set_value(count)
    finally:
        # close connection, remove subcsriptions, etc
        server.stop()
```

The opcua.json file:
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "localhost:4840/freeopcua/server/",
    "scanPeriodInMillis": 10000,
    "timeoutInMillis": 5000,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Device1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "attributes": [
          {
            "key": "temperature °C",
            "path": "${ns=2;i=5}"
          }
        ],
        "timeseries": [
          {
            "key": "humidity",
            "path": "${Root\\.Objects\\.Device1\\.TemperatureAndHumiditySensor\\.Humidity}"
          }
        ],
        "rpc_methods": [
          {
            "method": "multiply",
            "arguments": [2, 4]
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.Device1\\.serialNumber"
          }
        ]
      }
    ]
  }
}
```

I am not sure if I am doing something wrong or it is a bug.

I appreciate very much any help.

zbeacon, 2021-04-29:
Hi @jonathanwvd ,

Thank you for your interest in ThingsBoard IoT gateway. 
Please try to add parameter `"showMap": true, ` into your server section in opcua json configuration file  and check logs, the gateway will show pathes to nodes in logs. Also please try to use `Root.Objects.Device1` as a deviceNdePattern.

zbeacon, 2021-04-29:
I found the bug, and solve it in the master branch, please try to use it and in your opcua server please save *serialNumber* into the Device1 object instead of Root.Objects

jonathanwvd, 2021-04-29:
Thank you very much for the answer.
Now I install the latest version from gitlab. It is connecting and no error is shown, but I am still not getting any data and the nodes are not shown. 
When you ask me to change the save serialNumber, you mean to change this line?
`    name = server.nodes.objects.add_variable(idx, "serialNumber", "TEST")
`

zbeacon, 2021-04-29:
Yes, I mean please try to use ```name = device.add_variable(idx, "serialNumber", "TEST")``` instead of ```name = server.nodes.objects.add_variable(idx, "serialNumber", "TEST")```

jonathanwvd, 2021-04-29:
It is still not working. 
Actually a different device appears at thingsboard. 
It is strange that the only output I get from the gateway is this:
```
""2021-04-29 07:56:39" - INFO - [tb_gateway_service.py] - tb_gateway_service - 73 - Gateway starting..."
""2021-04-29 07:56:39" - INFO - [tb_gateway_service.py] - tb_gateway_service - 78 - ThingsBoard IoT gateway version: 2.6"
""2021-04-29 07:56:39" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 1"
""2021-04-29 07:56:39" - INFO - [tb_loader.py] - tb_loader - 66 - Import OpcUaConnector from /home/jonat/.local/share/virtualenvs/heron-aowqROaD/lib/python3.8/site-packages/thingsboard_gateway-2.6-py3.8.egg/thingsboard_gateway/connectors/opcua."
""2021-04-29 07:56:39" - INFO - [opcua_connector.py] - opcua_connector - 99 - Starting OPC-UA Connector"
""2021-04-29 07:56:39" - INFO - [tb_gateway_service.py] - tb_gateway_service - 136 - Gateway started."
""2021-04-29 07:56:39" - INFO - [opcua_connector.py] - opcua_connector - 124 - OPC-UA connector OPC-UA Connector connected to server localhost:4840/freeopcua/server/"
""2021-04-29 07:56:40" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
""2021-04-29 07:56:42" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
```

jonathanwvd, 2021-05-04:
Still not able to get data using OPC-UA. I have been trying the docker version again, but I am still unable to connect with the OPC server. Do I need to use a different address?
`ERROR - [opcua_connector.py] - opcua_connector - 110 - Connection refused on connection to OPC-UA server with url opc.tcp://admin@127.0.0.1:4840/freeopcua/server/"`

andrkass, 2021-05-07:
HI,
I have the same error since the new version. I updated last week and since then no data.

zbeacon, 2021-05-13:
Please try to use the version from the master branch.

Valentin-Schuele, 2021-05-20:
Hi,
I have the same issue. I only get one value displayed on my telemetry data. I cloned the current master branch and installed it from there.

jonathanwvd, 2021-05-27:
I was able to get data, but I need to have an attribute pointing to the same node as the time series. If I have just the attribute, it works fine, but if I have just the time series, I get the following error: 
```
""2021-05-27 13:13:11" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "Device TEST3", "deviceType": "default", "attributes": [], "telemetry": [{"temperature": "32.700000000000195"}]}"
""2021-05-27 13:13:11" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from OPC-UA Connector connector is invalid."
```

My json file that works
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "localhost:4840/freeopcua/server/",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions":false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Device1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "attributes": [
          {
            "key": "Humidity",
            "path": "${ns=2;i=4}"
          }
        ],
        "timeseries": [
          {
            "key": "temperature",
            "path": "${ns=2;i=4}"
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.Device1\\.serialNumber"
          }
        ]
      }
    ]
  }
}
```

jainabhishekindia, 2021-06-14:
Hi,

I am using below for opcu.json:

{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "xxxxxxxxxx:4840/",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions":false,
    "subCheckPeriodInMillis": 100,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.Device1",
        "deviceNamePattern": "Device ${Root\\.Objects\\.Device1\\.serialNumber}",
        "attributes": [
          {
            "key": "Humidity",
            "path": "${ns=2;i=5}"
          }
        ],
        "timeseries": [
          {
            "key": "Temperature",
            "path": "${ns=2;i=4}"
          }
        ],
        "attributes_updates": [
          {
            "attributeOnThingsBoard": "deviceName",
            "attributeOnDevice": "Root\\.Objects\\.Device1\\.serialNumber"
          }
        ]
      }
    ]
  }
}

===============================================================================================
And I am using below as my opcuaServer:

import sys
import time

from opcua import ua, Server

if __name__ == "__main__":

    # setup our server
    server = Server()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/")

    # setup our own namespace, not really necessary but should as spec
    name = "OPC_SIMULATION_SERVER"
    idx = server.register_namespace(name)

    device = server.nodes.objects.add_object(idx, "Device1")
    name = server.nodes.objects.add_variable(idx, "serialNumber", "OPCUADevice")
    name.set_writable()
    temperature_and_humidity = device.add_object(idx, "TemperatureAndHumiditySensor")
    temperature = temperature_and_humidity.add_variable(idx, "Temperature", 56.7)
    humidity = temperature_and_humidity.add_variable(idx, "Humidity", 68.7)

    # starting!
    server.start()

    try:
        count = 0
        while True:
            time.sleep(1)
            count += 0.1
            temperature.set_value(count)
    finally:
        # close connection, remove subcsriptions, etc
        server.stop()

===================================================================================================

I have tried to mimic the same config as done by Johnathan but I am getting below WARNING:

""2021-06-14 17:56:25" - WARNING - [opcua_connector.py] - opcua_connector - 411 - Device name node - not found, skipping device..."
""2021-06-14 17:56:25" - INFO - [opcua_connector.py] - opcua_connector - 133 - Reconnected to the OPC-UA server - 34.227.175.179:4840/"

I also don't see the device getting created.

I followed instructions on https://thingsboard.io/docs/iot-gateway/install/source-installation/ link for installation of iot gateway.
Can anyone please help.

samson0v, 2021-11-15:
Hi @jonathanwvd and @jainabhishekindia , try to use OPC-UA node paths configuration following to this example:

[image]
[image]

Thanks for your interest in ThingsBoard IoT Gateway!

samson0v, 2021-11-15:
@jonathanwvd if you want to use Gateway under the docker container, please follow this [instruction](https://github.com/thingsboard/thingsboard-gateway/issues/615).
This bug was fixed but the docker container has not been updated on the docker hub yet.

samson0v, 2021-11-29:
Hi @jonathanwvd, did it help, can we close the issue?

closed 2021-12-07 by zbeacon