thingsboard/thingsboard-gateway #511: [BUG] lastActivity ovewritten for all devices

santiago-ruiz, 2021-04-29:
**Describe the bug**
I'm still having the issue posted in the Thingsboard repo (issue [#3915](https://github.com/thingsboard/thingsboard/issues/3915)).  I've been trying different configurations but none seem to work correctly. In the cloud, the gateway has `Is Gateway` checked and `Overwrite activity time for connected devices` not checked. What I'm seeing is that every 30 seconds all the devices (including disconnected ones) have their ` lastActivityTime` rewritten. I encounter this bug in Gateway version 2.5.5.2 and in 2.6. I understand that this bug was already fixed so maybe it's just a configuration error. Thanks!

**Connector name:**
Modbus serial connector. Config file:
```json
{ "server":
   {
    "byteOrder": "BIG",
    "devices": [
      {
        "attributes": [],
        "timeseries": [
          {
            "address": 40,
            "functionCode": 3,
            "objectsCount": 1,
            "tag": "status_1",
            "type": "16uint"
          },
          {
            "address": 42,
            "functionCode": 3,
            "objectsCount": 1,
            "tag": "status_2",
            "type": "16uint"
          }
        ],
        "attributesPollPeriod": 3000,
        "deviceName": "device1",
        "sendDataOnlyOnChange": true,
        "timeseriesPollPeriod": 8000,
        "unitId": 1
      },
      {
        "attributes": [],
        "timeseries": [
          {
            "address": 40,
            "functionCode": 3,
            "objectsCount": 1,
            "tag": "status_1",
            "type": "16uint"
          },
          {
            "address": 42,
            "functionCode": 3,
            "objectsCount": 1,
            "tag": "status_2",
            "type": "16uint"
          }
        ],
        "attributesPollPeriod": 3000,
        "deviceName": "device2",
        "sendDataOnlyOnChange": true,
        "timeseriesPollPeriod": 8000,
        "unitId": 2
      }
    ],
    "method": "rtu",
    "port": "/dev/serial0",
    "baudrate": 9600,
    "stopbits": 1,
    "bytesize": 8,
    "parity": "E",
    "strict": true,
    "timeout": 2,
    "type": "serial"
  }
}
```
Gateway config file:
```yaml
thingsboard:
  host: thingsboard.cloud
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  security:
    accessToken: ACCESS_TOKEN
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
connectors:
  -
    name: Modbus Connector
    type: modbus
    configuration: modbus_serial.json
```

**Versions:**
 - OS: Ubuntu 18.04
 - Thingsboard IoT Gateway version 2.6-1
 - Python version 3.7

zbeacon, 2021-05-11:
Hi @santiago-ruiz ,

Thank you for your tests, you are right, there was one more issue with the gateway, and it updates devices activity. Please try the version form the master branch.

santiago-ruiz, 2021-05-14:
Hi @zbeacon,
I will be trying this fix in the following days. Thanks!

closed 2021-08-04 by zbeacon