thingsboard/thingsboard-gateway #512: [BUG] One-Way MQTT SSL not working

santiago-ruiz, 2021-04-30:
**Describe the bug**
I'm trying to configure Thingsboard IoT Gateway to work with One-Way SSL as explained [here](https://thingsboard.io/docs/iot-gateway/configuration). Configuration YAML looks like this:
```yaml
security:
    accessToken: PUT_YOUR_GW_ACCESS_TOKEN_HERE
    caCert: /etc/thingsboard-gateway/tb-cloud-chain.pem
```
I've downloaded the CA certificate as explained [here](https://thingsboard.io/docs/paas/user-guide/access-token/). I'm getting an error as shown down. I've tried changing the owner and group of the file to `thingsboard_gateway` but the error didn't change, so I'm guessing is not a permission problem.

**Connector name:**
Modbus serial.

**Error traceback:**
```
""2021-04-30 13:55:29" - ERROR - [tb_client.py] - tb_client - 134 - [Errno 0] Error"
Traceback (most recent call last):
File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_client.py", line 130, in run
min_reconnect_delay=self.__min_reconnect_delay)
File "/usr/lib/python3/dist-packages/thingsboard_gateway/tb_client/tb_device_mqtt.py", line 167, in connect
self._client.connect(self.__host, self.__port, keepalive=keepalive)
File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 839, in connect
return self.reconnect()
File "/usr/lib/python3/dist-packages/paho/mqtt/client.py", line 994, in reconnect
sock.do_handshake()
File "/usr/lib/python3.7/ssl.py", line 1117, in do_handshake
self._sslobj.do_handshake()
OSError: [Errno 0] Error
```

**Versions:**
 - OS: Raspberry Pi OS 5.10
 - Thingsboard IoT Gateway version 2.6-1
 - Python version 3.7

zbeacon, 2021-05-14:
Hi @santiago-ruiz ,

Thank you for your interest in ThingsBoard IoT gateway, please provide your yaml configuration file or check is the port in thingsboard section equals to 8883? It is used for secure mqtt connection.

santiago-ruiz, 2021-05-14:
Hi @zbeacon ,
Thanks for the answer. Yes, the problem was that the configuration port, that was set to the default one (1883). Now it's working perfectly. I think that setting the MQTT port to 8883 in order to configure SSL should be written somewhere in the documentation. Thanks!

closed 2021-05-14 by santiago-ruiz