thingsboard/thingsboard-gateway #519: Rest_connector: on_attributes_update crashes [BUG]

danielechermisi, 2021-05-10:
**Describe the bug**
When updating an attribute of a device connected to the gateway using rest_connector, the gateway raises an exception.

**Connector name :**
[REST Connector]

**Error traceback (If available):**
```
Exception in thread Attribute request to http://127.0.0.1:8080/sensor/Device SN-005/test:
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\threading.py", line 954, in _bootstrap_inner
    self.run()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.1520.0_x64__qbz5n2kfra8p0\lib\threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
TypeError: __send_request() missing 1 required positional argument: 'request'
```

**Versions (please complete the following information):**
 - OS: [e.g. Windows 10]
 - Thingsboard IoT Gateway version [e.g. 2.6]
 - Python version[e.g. 3.9.3]

danielechermisi, 2021-05-11:
Hi @zbeacon 
Please note that the issue occours when a shared attribute of a device is changed, so the parsing of the rpc request should not be relate to the configuration. Anyway, you can find below the configuration files used. 

We also noticed that in the on_attributes_update method in the rest_connector.py file, the Thread class is created with only 3 arguments
`
Thread(target=self.__send_request, args=(request_dict, response_queue, log),
`
while a further parameter of flask_request type is expected, i.e.
`
Thread(target=self.__send_request, args=(request_dict, response_queue, log, flask_request),
`
Thanks

Configuration files

` thingsboard:
  host: thingsboard.cloud
  port: 1883
  remoteShell: true
  remoteConfiguration: false
  security:
    accessToken: xxxxxxxxxxx
#    caCert: "./config/mqttserver.pub.pem"
  qos: 1
storage:
  type: memory
  read_records_count: 10
  max_records_count: 1000
connectors:
 -
   name: REST Connector
   type: rest
   configuration: rest.json
`
this is the rests.json

`
{
  "name": "REST Connector",
  "host": "0.0.0.0",
  "port": "5050",
  "mapping":[
    {
      "endpoint": "/test_device",
      "HTTPMethods": [
        "POST"
      ],
      "security":
      {
        "type": "anonymous"
      },
      "converter": {
        "type": "json",
        "deviceNameExpression": "Device ${name}",
        "deviceTypeExpression": "default",
        "attributes": [
          {
            "type": "string",
            "key": "model",
            "value": "${sensorModel}"
          }
        ],
        "timeseries": [
          {
            "type": "double",
            "key": "temperature",
            "value": "${temp}"
          },
          {
            "type": "double",
            "key": "humidity",
            "value": "${hum}"
          }
        ]
      }
    }
  ],
  "attributeUpdates": [
      {
        "HTTPMethod": "POST",
        "SSLVerify": false,
        "httpHeaders": {
          "CONTENT-TYPE": "application/json"
        },
        "security": {
          "type": "anonymous"
        },
        "timeout": 0.5,
        "tries": 3,
        "allowRedirects": true,
        "deviceNameFilter": ".*REST$",
        "attributeFilter": "data",
        "requestUrlExpression": "http://127.0.0.1:8080/sensor/${deviceName}/${attributeKey}",
        "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
      },
      {
        "HTTPMethod": "POST",
        "SSLVerify": false,
        "httpHeaders": {
          "CONTENT-TYPE": "application/json"
        },
        "security": {
          "type": "anonymous"
        },
        "timeout": 5,
        "tries": 3,
        "allowRedirects": true,
        "deviceNameFilter": ".*",
        "attributeFilter": ".*",
        "requestUrlExpression": "http://127.0.0.1:8080/sensor/${deviceName}/${attributeKey}",
        "valueExpression": "{\"${attributeKey}\":\"${attributeValue}\"}"
      }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": ".*",
      "requestUrlExpression": "http://127.0.0.1:8080/${deviceName}",
      "responseTimeout": 1,
      "HTTPMethod": "GET",
      "valueExpression": "${params}",
      "timeout": 0.5,
      "tries": 3,
      "httpHeaders": {
        "Content-Type": "application/json"
      },
      "security": {
        "type": "anonymous"
      }
    },
    {
      "deviceNameFilter": ".*",
      "methodFilter": "no-reply",
      "requestUrlExpression": "sensor/${deviceName}/request/${methodName}/${requestId}",
      "HTTPMethod": "POST",
      "valueExpression": "${params}",
      "httpHeaders": {
        "Content-Type": "application/json"
      }
    }
  ]
}

`

zbeacon, 2021-05-13:
Hi @danielechermisi ,

Thank you for your participating. Fix for this issue was added to the master branch and it will be available in the next release. Please try to use the version from the master branch and let us know if something go wrong.

closed 2021-08-04 by zbeacon