thingsboard/thingsboard-gateway #526: [HELP] Data from Modbus Connector connector is invalid.

yeduy, 2021-06-10:
**Describe the issue**

Hello,

I was able to get data in old version with this config. Now I'm getting an error that I can't understand.

**Configuration** (Attach your configuration file)

`
{
	"server": {
		"type": "serial",
		"method": "rtu",
		"port": "/dev/ttyS1",
		"baudrate": 9600,
		"stopbits": 1,
		"bytesize": 8,
		"parity": "N",
		"timeout": 3,
		"byteOrder": "BIG",
		"devices": [
			{
				"unitId": 22,
				"deviceName": "B1ENJ22",
				"deviceType": "enerji-analizoru",
				"timeseriesPollPeriod": 5000,
				"sendDataOnlyOnChange": false,
				"timeseries": [
					{
						"tag": "PL1",
						"type": "32int",
						"functionCode": 3,
						"objectsCount": 2,
						"address": 20,
						"multiplier": 0.0001
					},
					{
						"tag": "PL2",
						"type": "32int",
						"functionCode": 3,
						"objectsCount": 2,
						"address": 22,
						"multiplier": 0.0001
					},
					{
						"tag": "PL3",
						"type": "32int",
						"functionCode": 3,
						"objectsCount": 2,
						"address": 24,
						"multiplier": 0.0001
					},
					{
						"tag": "L1COSFI",
						"type": "32int",
						"functionCode": 3,
						"objectsCount": 2,
						"address": 38,
						"multiplier": 0.001
					},
					{
						"tag": "L2COSFI",
						"type": "32int",
						"functionCode": 3,
						"objectsCount": 2,
						"address": 40,
						"multiplier": 0.001
					},
					{
						"tag": "L3COSFI",
[... 1638 lines cut ...]
""2021-06-10 16:53:36" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "B1ENJ24", "deviceType": "enerji-analizoru", "telemetry": [{"PL1": 0.0}, {"PL2": 0.0}, {"PL3": 0.0}, {"L1COSFI": 1.0}, {"L2COSFI": 1.0}, {"L3COSFI": 1.0}, {"P_Import": 0.0}, {"P_Export": 0.0}, {"Q_Induktif": 0.0}, {"Q_Kapasitif": 0.0}, {"S": 0.0}, {"Frekans": 50.050000000000004}, {"AE_Import": 437001.955}, {"AE_Export": 2.279}, {"Induktif_Reaktif": 495442.09}, {"Kapasitif_Reaktif": 127.19500000000001}], "attributes": []}"
""2021-06-10 16:53:36" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from Modbus Connector connector is invalid."
""2021-06-10 16:53:40" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "B1ENJ25", "deviceType": "enerji-analizoru", "telemetry": [{"PL2": 0.2848}, {"PL3": 0.1408}, {"L1COSFI": -1.0}, {"L2COSFI": 0.978}, {"L3COSFI": -0.9560000000000001}, {"P_Import": 5.0016}, {"P_Export": 0.0}, {"Q_Induktif": 0.0736}, {"Q_Kapasitif": -0.14400000000000002}, {"S": 9.6384}, {"Frekans": 50.04}, {"AE_Import": 93595.24}, {"AE_Export": 0.203}, {"Induktif_Reaktif": 2683.089}, {"Kapasitif_Reaktif": 995.172}], "attributes": []}"
""2021-06-10 16:53:40" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from Modbus Connector connector is invalid."
""2021-06-10 16:53:41" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "B1ENJ26", "deviceType": "enerji-analizoru", "telemetry": [{"PL1": 10.236}, {"PL2": 8.706000000000001}, {"PL3": 7.44}, {"L1COSFI": 0.999}, {"L2COSFI": 0.996}, {"L3COSFI": 0.993}, {"P_Import": 24.396}, {"P_Export": 0.0}, {"Q_Induktif": 2.658}, {"Q_Kapasitif": 0.0}, {"S": 27.183}, {"Frekans": 50.050000000000004}, {"AE_Import": 473753.558}, {"AE_Export": 5.134}, {"Induktif_Reaktif": 548556.811}, {"Kapasitif_Reaktif": 159.29}], "attributes": []}"
""2021-06-10 16:53:41" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from Modbus Connector connector is invalid."
""2021-06-10 16:53:42" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "B1ENJ27", "deviceType": "enerji-analizoru", "telemetry": [{"PL1": 0.0}, {"PL2": 0.0}, {"PL3": 0.0}, {"L1COSFI": 1.0}, {"L2COSFI": 1.0}, {"L3COSFI": 1.0}, {"P_Import": 0.0}, {"P_Export": 0.0}, {"Q_Induktif": 0.0}, {"Q_Kapasitif": 0.0}, {"S": 0.0}, {"Frekans": 50.04}, {"AE_Import": 1097618.009}, {"AE_Export": 79.09700000000001}, {"Induktif_Reaktif": 1024253.563}, {"Kapasitif_Reaktif": 57.796}], "attributes": []}"
""2021-06-10 16:53:42" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from Modbus Connector connector is invalid."
""2021-06-10 16:53:43" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "B1ENJ28", "deviceType": "enerji-analizoru", "telemetry": [{"PL1": 0.0}, {"PL2": 0.0}, {"PL3": 0.0}, {"L1COSFI": 1.0}, {"L2COSFI": 1.0}, {"L3COSFI": 1.0}, {"P_Import": 0.0}, {"P_Export": 0.0}, {"Q_Induktif": 0.0}, {"Q_Kapasitif": 0.0}, {"S": 0.0}, {"Frekans": 50.04}, {"AE_Import": 78474.373}, {"AE_Export": 0.0}, {"Induktif_Reaktif": 2981.811}, {"Kapasitif_Reaktif": 642.36}], "attributes": []}"
""2021-06-10 16:53:43" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from Modbus Connector connector is invalid."
""2021-06-10 16:53:44" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "B1ENJ29", "deviceType": "enerji-analizoru", "telemetry": [{"PL1": 0.0}, {"PL2": 0.0}, {"PL3": 0.0}, {"L1COSFI": 1.0}, {"L2COSFI": 1.0}, {"L3COSFI": 1.0}, {"P_Import": 0.0}, {"P_Export": 0.0}, {"Q_Induktif": 0.0}, {"Q_Kapasitif": 0.0}, {"S": 0.0}, {"Frekans": 50.04}, {"AE_Import": 227350.863}, {"AE_Export": 0.0}, {"Induktif_Reaktif": 11346.585000000001}, {"Kapasitif_Reaktif": 1002.2420000000001}], "attributes": []}"
""2021-06-10 16:53:44" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from Modbus Connector connector is invalid."
""2021-06-10 16:53:45" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "B1ENJ30", "deviceType": "enerji-analizoru", "telemetry": [{"PL1": 0.0}, {"PL2": 0.0}, {"PL3": 0.0}, {"L1COSFI": 1.0}, {"L2COSFI": 1.0}, {"L3COSFI": 1.0}, {"P_Import": 0.0}, {"P_Export": 0.0}, {"Q_Induktif": 0.0}, {"Q_Kapasitif": 0.0}, {"S": 0.0}, {"Frekans": 50.04}, {"AE_Import": 773969.707}, {"AE_Export": 4324.665}, {"Induktif_Reaktif": 790020.933}, {"Kapasitif_Reaktif": 4406.195}], "attributes": []}"
""2021-06-10 16:53:45" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from Modbus Connector connector is invalid."
""2021-06-10 16:53:45" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "B1ENJ31", "deviceType": "enerji-analizoru", "telemetry": [{"PL1": 0.0}, {"PL2": 0.0}, {"PL3": 0.0}, {"L1COSFI": 1.0}, {"L2COSFI": 1.0}, {"L3COSFI": 1.0}, {"P_Import": 0.0}, {"P_Export": 0.0}, {"Q_Induktif": 0.0}, {"Q_Kapasitif": 0.0}, {"S": 0.0}, {"Frekans": 50.050000000000004}, {"AE_Import": 782017.157}, {"AE_Export": 2.12}, {"Induktif_Reaktif": 218420.274}, {"Kapasitif_Reaktif": 10870.022}], "attributes": []}"
""2021-06-10 16:53:45" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from Modbus Connector connector is invalid."
""2021-06-10 16:53:46" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "B1ENJ32", "deviceType": "enerji-analizoru", "telemetry": [{"PL1": 0.621}, {"PL2": 6.531000000000001}, {"PL3": 0.28200000000000003}, {"L1COSFI": 0.995}, {"L2COSFI": -1.0}, {"L3COSFI": 0.796}, {"P_Import": 24.093}, {"P_Export": 0.0}, {"Q_Induktif": 0.153}, {"Q_Kapasitif": -0.195}, {"S": 24.75}, {"Frekans": 50.04}, {"AE_Import": 408384.389}, {"AE_Export": 455.098}, {"Induktif_Reaktif": 33114.481}, {"Kapasitif_Reaktif": 5598.758}], "attributes": []}"
""2021-06-10 16:53:46" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from Modbus Connector connector is invalid."
""2021-06-10 16:53:47" - ERROR - [tb_utility.py] - tb_utility - 68 - No telemetry and attributes in data: {"deviceName": "B1ENJ33", "deviceType": "enerji-analizoru", "telemetry": [{"PL1": 1.506}, {"PL2": 1.131}, {"PL3": 1.008}, {"L1COSFI": 0.988}, {"L2COSFI": -1.0}, {"L3COSFI": -0.994}, {"P_Import": 3.645}, {"P_Export": 0.0}, {"Q_Induktif": 0.264}, {"Q_Kapasitif": -0.186}, {"S": 4.872}, {"Frekans": 50.050000000000004}, {"AE_Import": 360097.727}, {"AE_Export": 510.802}, {"Induktif_Reaktif": 125492.045}, {"Kapasitif_Reaktif": 640.021}], "attributes": []}"
""2021-06-10 16:53:47" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 348 - Data from Modbus Connector connector is invalid."`

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.6]
 - Python version[e.g. 3.6]

yeduy, 2021-06-14:
I changed tb_utility.py with old version. I think there is a bug on that code with new version.

JoachimVeulemans, 2021-06-22:
@yeduy When I use the release version, everything works. When I compile the master branch, I get the same error with the SNMP connector. @zbeacon

JoachimVeulemans, 2021-06-22:
@zbeacon 

I get it working when replacing
```
if data.get("telemetry") is not None:
    for entry in data.get("telemetry"):
        if entry.get("ts") is not None and len(entry.get("values")) > 0:
            got_telemetry = True
            break
```

with

```
if data.get("telemetry") is not None:
    for entry in data.get("telemetry"):
        # if entry.get("ts") is not None and len(entry.get("values")) > 0:
        got_telemetry = True
        break
```

in tb_utility/tb_utility.py

The code expects a field "ts" and I currently don't have this in my config nor I see it in the documentation.

snmp.json:
```
"attributes": [],
"telemetry": [
{
	"key": "title",
	"method": "get",
	"oid": "1.3.6.1.4"
},
```

zbeacon, 2021-10-13:
Hi @JoachimVeulemans ,

This issue is fixed in the latest version and in the master branch. Please try to use it.