thingsboard/thingsboard-gateway #529: I can connect to OPCUA Server from Thingsboard Iot, but device not getting created. [HELP]

jainabhishekindia, 2021-06-14:
Hi,

I am using below for opcua.json:

{
"server": {
"name": "OPC-UA Default Server",
"url": "xxxxxxxxxx:4840/",
"timeoutInMillis": 5000,
"scanPeriodInMillis": 5000,
"disableSubscriptions":false,
"subCheckPeriodInMillis": 100,
"showMap": false,
"security": "Basic128Rsa15",
"identity": {
"type": "anonymous"
},
"mapping": [
{
"deviceNodePattern": "Root\.Objects\.Device1",
"deviceNamePattern": "Device ${Root\.Objects\.Device1\.serialNumber}",
"attributes": [
{
"key": "Humidity",
"path": "${ns=2;i=5}"
}
],
"timeseries": [
{
"key": "Temperature",
"path": "${ns=2;i=4}"
}
],
"attributes_updates": [
{
"attributeOnThingsBoard": "deviceName",
"attributeOnDevice": "Root\.Objects\.Device1\.serialNumber"
}
]
}
]
}
}

===============================================================================================
And I am using below as my opcuaServer:

import sys
import time

from opcua import ua, Server

if name == "main":

# setup our server
server = Server()
server.set_endpoint("opc.tcp://0.0.0.0:4840/")

# setup our own namespace, not really necessary but should as spec
name = "OPC_SIMULATION_SERVER"
idx = server.register_namespace(name)

device = server.nodes.objects.add_object(idx, "Device1")
name = server.nodes.objects.add_variable(idx, "serialNumber", "OPCUADevice")
name.set_writable()
temperature_and_humidity = device.add_object(idx, "TemperatureAndHumiditySensor")
temperature = temperature_and_humidity.add_variable(idx, "Temperature", 56.7)
humidity = temperature_and_humidity.add_variable(idx, "Humidity", 68.7)

# starting!
server.start()

try:
    count = 0
    while True:
        time.sleep(1)
        count += 0.1
        #temperature.set_value(count)
finally:
    # close connection, remove subcsriptions, etc
    server.stop()
================================================================================================

I am getting below WARNING:

""2021-06-14 17:56:25" - WARNING - [opcua_connector.py] - opcua_connector - 411 - Device name node - not found, skipping device..."
""2021-06-14 17:56:25" - INFO - [opcua_connector.py] - opcua_connector - 133 - Reconnected to the OPC-UA server - 34.227.175.179:4840/"

I also don't see the device getting created.

I followed instructions on https://thingsboard.io/docs/iot-gateway/install/source-installation/ link for installation of iot gateway.
Can anyone please help.

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.7]

zbeacon, 2021-06-22:
Hi @jainabhishekindia ,

it looks like "serialNumber" variable not in the "Device1" object.

jainabhishekindia, 2021-06-22:
I have below line in code:

name = server.nodes.objects.add_variable(idx, "serialNumber", "OPCUADevice")

jainabhishekindia, 2021-06-22:
Do I need to add object inside Device1. Something like below:

serialNumber= device.add_object(idx, "serialNumber")
serialNumber= serialNumber.add_variable(idx, "serialNumber", 11)

zbeacon, 2021-09-06:
Hi @jainabhishekindia ,

yes, you can do like you told. Please create a new issue, if you cannot solve your problem.

closed 2021-09-06 by zbeacon