thingsboard/thingsboard-gateway #530: NO_ACTIVE_CONNECTION while sending RPC calls to my device

itsoli91, 2021-06-14:
**Component**

 * Rule Engine

**Description**
I'm using RPC call request node to send RPC calls to my device which is thermostat. Sometime it works fine and sometimes I get error below. What's Wrong ? I also added `inactivityTimeout` server-side attr to device for 15 days but still get this error.

```
java.lang.RuntimeException: NO_ACTIVE_CONNECTION
	at org.thingsboard.rule.engine.rpc.TbSendRPCRequestNode.lambda$onMsg$0(TbSendRPCRequestNode.java:134)
	at org.thingsboard.server.service.rpc.DefaultTbRuleEngineRpcService.lambda$sendRpcRequestToDevice$0(DefaultTbRuleEngineRpcService.java:125)
	at org.thingsboard.server.service.rpc.DefaultTbRuleEngineRpcService.lambda$processRpcResponseFromDevice$1(DefaultTbRuleEngineRpcService.java:150)
	at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:829)
```

Transformation Script :

```
var msg = {
    method: metadata.cs_location + "/" + metadata.originatorName + "/ThermostatMode/set",
    params: 3
};
metadata.oneway = true;
return {
    msg: msg,
    metadata: metadata,
    msgType: msgType
};
```

Gateway mqtt.json

```
{
  "broker": {
    "name": "Default Local Broker",
    "host": "mosquitto",
    "port": 1883,
    "clientId": "ThingsBoard_gateway",
    "security": {
      "type": "basic",
      "username": "xx",
      "password": "xx"
    }
  },
  "mapping": [// my mappings ],
  "connectRequests": [
    {
      "topicFilter": "xx/Building/+/+/nodeinfo",
      "deviceNameJsonExpression": "${name}"
    }
  ],
  "serverSideRpc": [
    {
      "deviceNameFilter": ".*",
      "methodFilter": ".*",
      "requestTopicExpression": "xx/${methodName}",
      "valueExpression": "${params}"
    }
  ],
  "disconnectRequests": [],
  "attributeUpdates": []
}
```

**Environment**

 * OS:  Linux Ubuntu
 * ThingsBoard: latest

zbeacon, 2021-06-22:
Hi @itsoli91 ,

Please try to change the inactivityTimeout for the gateway device.

itsoli91, 2021-06-22:
Hi,
Still dose not work, now I'm pretty sure that ThingsBoard cant send RPC Request while using gateway. I'm not sure but I checked gateway logs it works fine for reading data from devices and sending to ThingsBoard but while calling RPC call ThingsBoard cant connect to gateway and I cant see any log for sending RPC calls. I set time out to 120 but not working.

slibutti, 2021-06-24:
Hi there,

I suggest also changing TB_TRANSPORT_SESSIONS_INACTIVITY_TIMEOUT in the ThingsBoard configuration.
Make it higher than the device transmission period, at least.

S

samson0v, 2021-11-16:
Hi @itsoli91, is your problem actual for today?

closed 2021-11-29 by zbeacon