thingsboard/thingsboard-gateway #533: IndexError: list index out of range

keyiis, 2021-06-28:
I used gateway to connect opcua server,but always unsuccessful,
below is gateway config
```
{
  "server": {
    "name": "OPC-UA Default Server",
    "url": "192.168.8.166:4334/UA/MyLittleServer/",
    "timeoutInMillis": 5000,
    "scanPeriodInMillis": 5000,
    "disableSubscriptions":true,
    "subCheckPeriodInMillis": 0,
    "showMap": false,
    "security": "Basic128Rsa15",
    "identity": {
      "type": "anonymous"
    },
    "mapping": [
      {
        "deviceNodePattern": "Root\\.Objects\\.mypc",
        "deviceNamePattern": "${Root\\.Objects\\.mypc\\.Name}",
        "attributes": [

        ],
        "timeseries": [
         
        ],
        "rpc_methods": [
          
        ],
        "attributes_updates": [
         
        ]
      }
    ]
  }
}
```
my opcua server already runing
[image]
but gateway get error
```
IndexError: list index out of range

""2021-06-28 03:11:37" - DEBUG - [opcua_connector.py] - opcua_connector - 284 - [{'Root\\.Objects\\.mypc': {'deviceNodePattern': 'Root\\.Objects\\.mypc', 'deviceNamePattern': '${Root\\.Objects\\.mypc\\.Name}', 'attributes': [], 'timeseries': [], 'rpc_methods': [], 'attributes_updates': []}}]"

""2021-06-28 03:11:42" - DEBUG - [opcua_connector.py] - opcua_connector - 396 - Looking for device name"

""2021-06-28 03:11:42" - ERROR - [opcua_connector.py] - opcua_connector - 534 - can only concatenate str (not "bytes") to str"

Traceback (most recent call last):

File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-2.6-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 493, in __search_node

new_node_path = self.get_node_path(child_node)

File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-2.6-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 459, in get_node_path

return 'Root.Objects.' + ID

TypeError: can only concatenate str (not "bytes") to str

""2021-06-28 03:11:42" - ERROR - [opcua_connector.py] - opcua_connector - 283 - list index out of range"

Traceback (most recent call last):

File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-2.6-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 266, in scan_nodes_from_config

devices_info_array = self.__search_general_info(device_configuration)

File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-2.6-py3.9.egg/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 408, in __search_general_info

device_name_node = device_name_node[0]

IndexError: list index out of range

""2021-06-28 03:11:42" - DEBUG - [opcua_connector.py] - opcua_connector - 284 - [{'Root\\.Objects\\.mypc': {'deviceNodePattern': 'Root\\.Objects\\.mypc', 'deviceNamePattern': '${Root\\.Objects\\.mypc\\.Name}', 'attributes': [], 'timeseries': [], 'rpc_methods': [], 'attributes_updates': []}}]"
```

zhuyujie1993, 2021-08-21:
fork

samson0v, 2021-09-06:
Hi @keyiis, please update your Gateway to the newest version via the master branch, if it does not help, please provide new info.
Thanks for your interest in ThingsBoard IoT Gateway!