thingsboard/thingsboard-gateway #535: [HELP] bacnet relinquish command

vac-pav, 2021-07-12:
I use bacnet connector which seems to work quite well. I can read and write properties at individual priorities using RPC. But I cannot find way how to relinquish already written value. E.g value written at priority 8 (manual control) and I need to relinquish this value.

Is there any way to do it with thingsboard or is this feature not implemented?

samson0v, 2021-11-16:
Hi @vac-pav and @WeiTangLau, we will add this functionality to the Gateway.

Thanks for your interest in ThingsBoard IoT Gateway.

samson0v, 2021-11-17:
Hi @vac-pav and @WeiTangLau, after investigation, we discover that if you overwrite the priority value below, then it is not possible to increase it because it will not be written. This is a limitation of BACNet.

Sorry for the inconvenience.

vac-pav, 2021-11-17:
Hi @samson0v , thanks for reply. Do you mean that it is not possible at priority 8 or at all? You are probably right, that at priority 8 this is not possible as this is manual control, this was probably my fault. But what about if we use priority 9. Then it should be possible to write there null value to remove command at priority 9. Or am I wrong? Is this supported by current version or can you implement it?

Kretiss, 2022-02-21:
Hi @samson0v 

How is it with using priorities in the BACnet connector? Is there any way to set the priority in the RPC request? I can't find any info about it in the documentation. Is this feature already implemented?

Thank you for answer.

NistantriTech-Dhaval, 2024-10-16:
@samson0v @vac-pav 
did you guys got any alternative? I also need to release the command written at priority but it seems no option there

samson0v, 2024-12-03:
Hi @vac-pav and everyone!
We are pleased to inform you that we added a new Async BACnet connector that used bacpypes3 (via PR #1607).
The old BACnet connector is now deprecated and can only be enabled manually.
Please, run the latest version of the gateway via the master branch and let us know about the results.

imbeacon, 2025-07-08:
Hi,

The new version is released. Please try to use it, feel free to open new issue if you discover something.