thingsboard/thingsboard-gateway #537: [HELP]  ERROR - [modbus_connector.py] - modbus_connector - 185 - Connection lost! Reconnecting..." gateway

redasdasd, 2021-07-15:
**thingsboard: 3.2.1 gateway: 2.3**
if i writed  "timeseriesPollPeriod": 120000 in Modbus.json. No error occurred
if i writed  "timeseriesPollPeriod": 600000 in Modbus.json. An error occurred： ERROR - [modbus_connector.py] - modbus_connector - 185 - Connection lost! Reconnecting...

**thingsboard: 3.2.1 gateway: 2.6**
if i writed  "timeseriesPollPeriod": 120000 in Modbus.json. No error occurred
if i writed  "timeseriesPollPeriod": 600000 in Modbus.json. An error occurred： ERROR - [bytes_modbus_uplink_converter.py] - bytes_modbus_uplink_converter - 82 - Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message receiv
raspberrypi python3[10516]: NoneType: None

Why does this error occur? How can i fix this error？

redasdasd, 2021-11-01:
I have checked these documents last week. The file I found is as shown below:
raspberrypi4:/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/modbus $ ls bytes_modbus_downlink_converter.py  bytes_modbus_uplink_converter.py  __init__.py  modbus_connector.py  modbus_converter.py  __pycache__

zbeacon, 2021-11-01:
Hi @redasdasd ,

The new version of the gateway is available, please update your instance and try again.

closed 2021-11-19 by zbeacon