thingsboard/thingsboard-gateway #543: [BUG] Example mqtt.json config file is missing attributeRequests section

pbrkr, 2021-07-27:
The example MQTT configuration in thingsboard_gateway/config/mqtt.json is missing the `attributeRequests` section required by the MQTT Connector.

zbeacon, 2021-08-04:
Hi @pbrkr ,

Thank you for your interest in ThingsBoard IoT gateway.
`attributeRequest` section was presented in the configuration at the very beginning of the project development, at the moment it is renamed to `attributeUpdates`. But it was left in the code to keep the backward compatibility.

pbrkr, 2021-08-05:
@zbeacon please re-open this issue.

The code in thingsboard_gateway/connectors/mqtt/mqtt_connector.py still requires `attributeRequests` to be present in the config. If it is not present, an error is printed (see call to `load_handlers()` on line 73, error logged on line 135 of current master). If this section is just kept for backwards compatibility then no error should be printed if the `attributeRequests` section is missing. Perhaps `load_handlers()` needs an optional flag?

zbeacon, 2021-08-05:
I agree with you, we will remove the error logging in this case.

samson0v, 2021-08-06:
Hi @pbrkr, fixed your issue, please, update your gateway to the latest version via the master branch.

closed 2021-09-06 by zbeacon