thingsboard/thingsboard-gateway #544: [BUG] MQTT converter fails if data itself is an array of objects

rduivenvoorde, 2021-07-27:
**Describe the bug**
Not able to retrieve timeseries value if the data/message is a list itself

The data the gateway received is something like: ```[{"id":"123456"}]``` (note: the message is actually a `docker inspect` result)
Using the MQTT converter, I seem to be able to retrieve the value by using ```"${ [0].id }"```
But I receive the traceback below, I think because:
https://github.com/thingsboard/thingsboard-gateway/blob/master/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py#L72
tries to find a ts or timestamp key in the (straight) data:
```
 if datatype == 'timeseries' and (data.get("ts") is not None or data.get("timestamp") is not None):
```
but as said: data is an array itself so a call to ```data.get(...)``` will result in the error below.

A solution would maybe be to check if data is a list, and if not do current lines, else either take only the first one, or use a json expression?

**Connector name (If bug in the some connector):**
MQTT Connector

**Error traceback (If available):**
```
""2021-07-27 14:48:00" - ERROR - [json_mqtt_uplink_converter.py] - json_mqtt_uplink_converter - 78 - 'list' object has no attribute 'get'"
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/thingsboard_gateway-2.6-py3.9.egg/thingsboard_gateway/connectors/mqtt/json_mqtt_uplink_converter.py", line 72, in convert
    if datatype == 'timeseries' and (data.get("ts") is not None or data.get("timestamp") is not None):
AttributeError: 'list' object has no attribute 'get'
****
```

**Versions (please complete the following information):**
 - OS: Default Docker image

samson0v, 2021-08-02:
Hi Richard Duivenvoorde, fixed the issue, please, update your gateway to the newest version via the master branch and let us know if it helps you.

rduivenvoorde, 2021-08-02:
@samson0v Thanks! Will do after my holidays when I'm back at work and report back. But thanks for the swift reaction!

closed 2021-09-06 by zbeacon