thingsboard/thingsboard-gateway #547: [BUG] Automatic update is undocumented and cannot be disabled

pbrkr, 2021-08-05:
Reviewing the log messages from thingsboard-gateway I saw the following:

```
Aug 05 17:54:13 bbe python3[21414]: ""2021-08-05 17:54:13" - DEBUG - [tb_updater.py] - tb_updater - 69 - Checking for new version"
```

This led me to look at thingsboard_gateway/gateway/tb_updater.py and the code in thingsboard_gateway/gateway/tb_gateway_service.py which creates a TBUpdater object. This appears to be checking for updates to the thingsboard-gateway package and installing them using `pip` when available.

This is not a feature we want when using thingsboard-gateway. We want to be able to manage updates centrally and have consistent versions of software on each gateway device to ease debugging any issues which occur. This feature should be off by default and enabled by a key in the config file by users who want automatic updates.

It also appears that this feature is not covered in the documentation - I think that's a serious omission as it can lead to a device running a different version of the software to the version which is installed. We have a couple of patches applied to thingsboard-gateway when we build our rootfs images - these patches would be removed by such an automatic update.

zbeacon, 2021-08-06:
Hi @pbrkr ,

Thank you for your interest in ThingsBoard IoT gateway. This feature just check is update available for the current gateway instance and update the gateway only in case of receiving  RPC method "update" from ThingsBoard. you are right this feature is not documented yet, we will add information about this feature to our documentation.

closed 2021-09-06 by zbeacon

pbrkr, 2021-09-08:
Hi @zbeacon, if the intention is to resolve these concerns, why has this issue been closed? If you intend to resolve these concerns I would expect that the issue be left open. Otherwise, where is this defect tracked?