thingsboard/thingsboard-gateway #555: [BUG]

zhuyujie1993, 2021-08-23:
**Describe the bug**
Clear and concise description of what the bug is.

**Connector name (If bug in the some connector):**
[e.g. OPC-UA Connector]

**Error traceback (If available):**
```
'deviceType':error
{'deviceName': 'opcua_sensor01', 'deviceType': 'ns=2;i=3', 'attributes': [], 'telemetry': [{'BatteryLevel': '71.20000000000023'}

'deviceType': 'ns=2;i=3'？？？？
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

samson0v, 2021-09-02:
Hi @zhuyujie1993, please provide more information about your Gateway configuration files (config/tb_gateway.yaml, config/opcua.json).
Thanks for your interest in ThingsBoard IoT Gateway!

zhuyujie1993, 2021-09-03:
opcua.json: 'deviceTypePattern'  Could not be specified successfully

tb_gateway.yaml

zbeacon, 2021-09-03:
Hi @zhuyujie1993 ,

Thank you for your interest  in ThingsBoard IoT gateway, but unfortunately,  we cannot help you with your issue because of you didn't provide us requested information (content of opcua.json file and any information  about the opcua server you used). We will close this issue, if you will be unable to solve the issue  by changing configuration - please open the new issue, but provide at very beginning  of the issue content of your configuration files.

closed 2021-09-03 by zbeacon