thingsboard/thingsboard-gateway #558: [HELP] 'TBGatewayService' object has no attribute 'connectors_configs'

nguyenlam185, 2021-08-29:
Hi guys,
I am currently working with Thingsboard IoT Gateway and trying to get the RPC remote shell to work.
The Thingsboard IoT Gateway has been installed on a Raspberry Pi4 which's running on Ubuntu Server 20.04.
My tb_gateway.yaml looks like below:

```
  host: 192.168.0.162
  port: 1883
  remoteShell: true
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: .....
  qos: 1
storage:
  type: file
  data_folder_path: ./data/
  max_file_count: 10
  max_read_records_count: 10
  max_records_per_file: 10000
connectors:
  -
    name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

```
I think the Gateway is able to connect to Thingsboard Server whichis hosted on another computer as the server attributes *active* value is *true*
Now, when I tried to setup the RPC Remote shell, it seems like it failed. It cannot get the OS information, and when I try to enter some commands, the things I got is a message about *connectors_configs*

```
Welcome to ThingsBoard RPC remote shell.
Current target device for RPC terminal: TB_Gateway
Remote platform info:
OS: undefined
TB_Gateway> ls
'TBGatewayService' object has no attribute 'connectors_configs'
TB_Gateway> 
 ```
Below is the lines I got when I check the status of thingsboard service

```
ubuntu@ubuntu:~$ systemctl status thingsboard-gateway
● thingsboard-gateway.service - ThingsBoard Gateway
     Loaded: loaded (/etc/systemd/system/thingsboard-gateway.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2021-08-29 08:03:56 UTC; 10min ago
   Main PID: 1998 (python3)
      Tasks: 5 (limit: 9257)
     CGroup: /system.slice/thingsboard-gateway.service
             └─1998 /usr/bin/python3 -c from thingsboard_gateway.tb_gateway import daemon; daemon()

Aug 29 08:11:55 ubuntu python3[1998]: ""2021-08-29 08:11:55" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 560 - 'TBGatewayService' object has no attribute 'connectors_configs'"
Aug 29 08:11:55 ubuntu python3[1998]: Traceback (most recent call last):
Aug 29 08:11:55 ubuntu python3[1998]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 537, in _rpc_request_handler
Aug 29 08:11:55 ubuntu python3[1998]:     if self.connectors_configs.get(module):
Aug 29 08:11:55 ubuntu python3[1998]: AttributeError: 'TBGatewayService' object has no attribute 'connectors_configs'
Aug 29 08:12:18 ubuntu python3[1998]: ""2021-08-29 08:12:18" - ERROR - [tb_gateway_service.py] - tb_gateway_service - 560 - 'TBGatewayService' object has no attribute 'connectors_configs'"
Aug 29 08:12:18 ubuntu python3[1998]: Traceback (most recent call last):
Aug 29 08:12:18 ubuntu python3[1998]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/gateway/tb_gateway_service.py", line 537, in _rpc_request_handler
Aug 29 08:12:18 ubuntu python3[1998]:     if self.connectors_configs.get(module):
Aug 29 08:12:18 ubuntu python3[1998]: AttributeError: 'TBGatewayService' object has no attribute 'connectors_configs'
ubuntu@ubuntu:~$

```
Thanks in advance for any suggestion to help me out of this situation.

zbeacon, 2021-09-06:
Hi @nguyenlam185 ,

Please try to use the version from the master branch.

nguyenlam185, 2021-09-16:
Hi @zbeacon ,
Thanks for your reply. 
I followed strictly the installation guide from this link [https://thingsboard.io/docs/iot-gateway/install/deb-installation/](https://thingsboard.io/docs/iot-gateway/install/deb-installation/)
So, I don't know the packages is on which branch, also I have not found any instruction on installing the software from git repository.
Could you please brief me how to install the version from master branch?
Thank you for your support

nguyenlam185, 2021-09-18:
Hi @zbeacon,
Just want to let you know that rolling back to the version 2.6 worked well now.
However, I always want to try to lasted version, so I would appreciate if you could help me out of the issue of v2.7

Best regards

samson0v, 2021-09-24:
Hi @nguyenlam185, if you want to install IoT Gateway from the master branch, you have to: 
1. clone the repository
2. `cd thingsboard-gateway`
3. create virtual environment `python3 -m venv venv`
4. activate the virtual environment `source venv/bin/activate`
5. `python setup.py install`
6. and finally run the Gateway `python thingsboard-gateway/tb-gateway.py`

nguyenlam185, 2021-10-04:
Hi @samson0v,
Thank you for your instruction. I have tried it and got the following log
```
""2021-10-04 09:23:48" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 336 - "MQTT Broker Connector" subscription success to topic /sensor/data, subscription message id = 29"
""2021-10-04 09:23:48" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 329 - ()"
""2021-10-04 09:23:48" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 336 - "MQTT Broker Connector" subscription success to topic /sensor/+/data, subscription message id = 30"
""2021-10-04 09:23:48" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 329 - ()"
""2021-10-04 09:23:48" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 336 - "MQTT Broker Connector" subscription success to topic /custom/sensors/+, subscription message id = 31"
""2021-10-04 09:23:49" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 329 - ()"
""2021-10-04 09:23:49" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 336 - "MQTT Broker Connector" subscription success to topic sensor/connect, subscription message id = 32"
""2021-10-04 09:23:49" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 329 - ()"
""2021-10-04 09:23:49" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 336 - "MQTT Broker Connector" subscription success to topic sensor/+/connect, subscription message id = 33"
""2021-10-04 09:23:49" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 329 - ()"
""2021-10-04 09:23:49" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 336 - "MQTT Broker Connector" subscription success to topic sensor/disconnect, subscription message id = 34"
""2021-10-04 09:23:49" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 329 - ()"
""2021-10-04 09:23:49" - |INFO| - [mqtt_connector.py] - mqtt_connector - _on_subscribe - 336 - "MQTT Broker Connector" subscription success to topic sensor/+/disconnect, subscription message id = 35"
Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/dnguyen/thingsboard-gateway/venv/lib/python3.8/site-packages/paho_mqtt-1.5.1-py3.8.egg/paho/mqtt/client.py", line 3452, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/home/dnguyen/thingsboard-gateway/venv/lib/python3.8/site-packages/paho_mqtt-1.5.1-py3.8.egg/paho/mqtt/client.py", line 1779, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/home/dnguyen/thingsboard-gateway/venv/lib/python3.8/site-packages/paho_mqtt-1.5.1-py3.8.egg/paho/mqtt/client.py", line 1181, in loop
    rc = self.loop_read(max_packets)
  File "/home/dnguyen/thingsboard-gateway/venv/lib/python3.8/site-packages/paho_mqtt-1.5.1-py3.8.egg/paho/mqtt/client.py", line 1572, in loop_read
    rc = self._packet_read()
  File "/home/dnguyen/thingsboard-gateway/venv/lib/python3.8/site-packages/paho_mqtt-1.5.1-py3.8.egg/paho/mqtt/client.py", line 2295, in _packet_read
    data = self._sock_recv(self._in_packet['to_process'])
  File "/home/dnguyen/thingsboard-gateway/venv/lib/python3.8/site-packages/paho_mqtt-1.5.1-py3.8.egg/paho/mqtt/client.py", line 664, in _sock_recv
    return self._sock.recv(bufsize)
AttributeError: 'NoneType' object has no attribute 'recv'
```
However, I installed the gateway (following the guideline from the documentation) on a fresh installed Ubuntu Raspberry worked well. Maybe it conflicted with something on my Raspberry at the beginning.
So, I think we can close this ticket

Thank you for your support

closed 2021-10-04 by nguyenlam185