thingsboard/thingsboard-gateway #563: [HELP] OTA Updates not working with gateway

feg624, 2021-09-02:
**Describe the issue**
We're evaluating Thingsboard for device management. Our objective is to install the Thingsboard gateway into an edge device, which then connects to the source devices. We're interested in the OTA Update feature to run on the gateway (to run a software update on the edge device)... we were able to open a RPC Shell from the dashboard into the gateway, but we haven't had luck with OTA Update. Question is: does the gateway have OTA Update feature out of the box? or do we need to code some script (adopting the SDK) in order to have that capability on the edge device (gateway)?

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**
```
thingsboard:
  host: mytb
  port: 1883
  remoteShell: true
  remoteConfiguration: true
  security:
    accessToken: xxx
#    caCert: "./config/mqttserver.pub.pem"
  qos: 1
```

**Connector name (If you need help with some connector/converter):**
N/A

**Error traceback (If it was raised):**
N/A

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04
 - Thingsboard IoT Gateway version 2.6
 - Python version 3.9

zbeacon, 2021-09-06:
Hi @feg624 ,

Thank you for your interest in ThingsBoard IoT gateway. At the moment the gateway doesn't support OTA updates, but we are going to add support of this feature to the gateway.

feg624, 2021-09-09:
Thanks @zbeacon for the clarification! One additional question though, do you know whether Thingsboard provides a solution that allows us to send OTA Updates to the device on which we have the gateway running? or to another device without the gateway involved?

I'm asking as when reading the docs and looking at the Thingsboard repos, it seems to me that we would need to develop the client software by adopting the SDK, which may not be a valid option for our use case. Do you know of any other "out of the box" client component that would allow the device to receive OTA Updates?

Thanks in advance,

Franco

samson0v, 2021-09-24:
Hi @feg624, you can use [Python SDK](https://github.com/thingsboard/thingsboard-python-client-sdk) or [Arduino SDK](https://github.com/thingsboard/thingsboard-arduino-sdk).

closed 2021-09-24 by feg624

eydonmlau, 2022-04-13:
> Hi @feg624 ,
> 
> Thank you for your interest in ThingsBoard IoT gateway. At the moment the gateway doesn't support OTA updates, but we are going to add support of this feature to the gateway.

Hi @imbeacon Does the thingboard gateway support ota update right now? 
Best regards.