thingsboard/thingsboard-gateway #566: [HELP] Is there any guide on how to setup development environment.

csm1sh, 2021-09-08:
**Describe the issue**
Report the below error when I try to run the source code.

C:\Users\jhu019\Downloads\thingsboard-gateway-master\thingsboard_gateway>python tb_gateway.py
Traceback (most recent call last):
  File "C:\Users\<HOME>\Downloads\thingsboard-gateway-master\thingsboard_gateway\tb_gateway.py", line 16, in <module>
    from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService
ModuleNotFoundError: No module named 'thingsboard_gateway'

**Versions (please complete the following information):**
 - OS: [e.g. window 10]
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.9]

samson0v, 2021-09-09:
Hi @csm1sh, you have to install all requirements using the next command:
```
python setup.py install
```

Thanks for your interest in ThingsBoard IoT Gateway.

closed 2021-10-13 by zbeacon

pgirgis, 2023-08-25:
The latest setup.py has a comma missing on line
urllib3<=1.26.15requests>=2.31.0
Once added works fine.