thingsboard/thingsboard-gateway #574: [HELP] Unable to get BACnet connector working

WeiTangLau, 2021-09-22:
**Describe the issue**
I am trying to connect to scrape the data from my devices using BACnet protocol. However, after I have done with the configurations, I am not able to get the data. Furthermore, my `connection.log` is empty. 

Also I am actually using BACnet IP device as an interface to BACnet MSTP devices. Will thingsboard be able to communicate with BACnet MSTP devices in this manner? Would be lovely if there are some configurations for me to follow!

**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

**Connector name (If you need help with some connector/converter):**
[e.g. BACnet Connector]

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [e.g. 2.7]
 - Python version[e.g. 3.8.10]

samson0v, 2021-09-24:
Hi @WeiTangLau, please change all log levels from INFO to DEBUG in config/logs.conf and provide the error logs from your terminal.

zbeacon, 2021-10-18:
Hi, @WeiTangLau ,

It looks like the issue is in configuration file. Please check that maxApduLengthAccepted parameter in your configuration json file is a number and not a string.

zbeacon, 2021-10-20:
@WeiTangLau ,

Now it looks like the gateway cannot find the BACnet device in the network:
It sends WhoIs request and doesn't receive IAm response. Please check the address of the device, is it correct. 
If it is correct please share your bacnet.json configuration file to ibarkov@thingsboard.io and give an access to this device (VPN or it's public IP), we will try to connect to it and configure the gateway to get data.

WeiTangLau, 2021-10-20:
Hi @zbeacon 

below is my `bacnet.json`

```
{
  "general": {
    "objectName": "TB_Gateway",
    "address": "192.168.1.80:47808",
    "objectIdentifier": 599,
    "maxApduLengthAccepted": 1024,
    "segmentationSupported": "segmentedBoth",
    "vendorIdentifier": 15
  },
  "devices": [
    {
      "deviceName": "BACnet Device ${objectName}",
      "deviceType": "default",
      "address": "192.168.1.100:47808",
      "pollPeriod": 10,
      "attributes": [
        {
          "key": "toggle",
          "objectId": "binaryValue:0",
          "propertyId": "presentValue"
        }
      ]
    }
  ]
}
```

Furthermore, here are the devices that are running on the network

[image]

We are currently running on TB Cloud and TB Edge. I not sure how to give you access.