thingsboard/thingsboard-gateway #575: Disconnected client, result code 0 [Thingsboard gateway-Python module]

santobedi, 2021-09-22:
**Describe the issue**
After updating the host and access token in tb_gateway.yaml file, I used  **sudo python3 -c 'from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService; TBGatewayService("./tb_gateway.yaml")'**  command to rerun the gateway. After running this command, I can see keys and values in the Latest telemetry section of Device details of web UI (updates only once when I run the above command). However, I can see nothing on the command prompt of my microcomputer. Pressing ctrl+c (perhaps its a keyboard interruption for MobaXterm), I see the following: 

**[STREAM ONLY] 2021-09-22 10:52:18,016 - DEBUG - [tb_device_mqtt.py] - tb_device_mqtt - 125 - Disconnected client: <paho.mqtt.client.Client object at 0xb5739db0>, user data: None, result code: 0**

The above lines are displayed when I run **thingsboard-gateway** command too. 

What does this **result code 0** mean? What is causing the **Disconnected client** ? 

Moreover, I notice that files/folders inside the 'thingsboard-gateway' directory given [here](https://github.com/thingsboard/thingsboard-gateway/tree/master/thingsboard_gateway)  do not match with the files/folders under the same directory (/etc/thingsboard-gateway/) in the microcomputer after installing things board gateway (only 'config' folder). 

Does it mean that the Thingsboard gateway is not installed completely?  

**Connector name (If you need help with some connector/converter):**
MQTT Connector

samson0v, 2021-09-24:
Hi @santobedi, when you run only thingsboard-gateway, that's run gateway as a terminal app but not as a background task (as a service), for this reason, when you logged out from your server, the terminal close all "foreground" task.
For more understanding example: if you run the VIM/Nano/Emacs in your terminal and logged out, the text editor will close.
Thanks for your interest in ThingsBoard IoT Gateway!

santobedi, 2021-09-24:
@samson0v  Thank you for the reply. My system is working properly now (after starting the gateway). However, I still wonder about the frequent 'Disconnected' debug msg (it happens after rerunning the gateway with python command and the 'thingsboard-gateway' command. Is it a normal behavior?

samson0v, 2021-09-24:
@santobedi no, it is very strange behaviour.

samson0v, 2021-11-29:
Hi @santobedi, did it help, can we close the issue?

santobedi, 2021-11-29:
Hello @samson0v ,
I've not used Version 2.9 yet. I hope it will solve the issue. It's fine to close the issue. Thank you.

closed 2021-12-03 by santobedi