thingsboard/thingsboard-gateway #576: How to send rpc call through gateway to device?

chenliangrui, 2021-09-22:
Device send data through gateway to thingsboard is fine.Now, I cant figure out how to send rpc call through gateway to device.
In my opinion, should send rpc call to gateway,and then gateway handle that ,and gateway create topic in external broker,so my device can subscribe like ‘sensor/SN-001/request/no-reply/1’ in order to recive message.If im right,how can i send rpc call to gateway.(use Control Widget's Send RPC button?)
And i had read official document，but im still struggling with this problem！
plz！！！

chenliangrui, 2021-09-23:
> Device send data through gateway to thingsboard is fine.Now, I cant figure out how to send rpc call through gateway to device.
> In my opinion, should send rpc call to gateway,and then gateway handle [...]

Emmmm...Now its fine,i just forget when i restart the btoker my device info be cleared and i need notice gateway about my device.
Thingsboard is perfect!

closed 2021-09-23 by chenliangrui

locole-dev, 2025-02-25:
How did you do that? If you can do it, please teaching me