thingsboard/thingsboard-gateway #578: [HELP]

RudolphAschmoneit, 2021-09-22:
**Describe the issue**
Hey at all. I try to get a thingsboard_gateway up and running, which describes to a mosquitto broker hosted on the same machine and piping it to a thingsboard also hosted on the same machine. Tb and tb_gw are setuped as deamons. I use the mqtt connector. The gateway could connect to the brooker and the gateway succesful, but it doesn't find the devices:

```
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [mqtt_connector.py] - mqtt_connector - 160 - Number of accepted attributeUpdates handlers: 0"
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [mqtt_connector.py] - mqtt_connector - 164 - Number of rejected attributeUpdates handlers: 0"
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [tb_gateway_service.py] - tb_gateway_service - 139 - Gateway started."
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [mqtt_connector.py] - mqtt_connector - 234 - MQTT Broker Connector connected to 127.0.0.1:1884 - successfufully."
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [tb_loader.py] - tb_loader - 66 - Import JsonMqttUplinkConverter from /usr/lib/python3/dist-packages/thingsboard_gateway/connectors/mqtt."
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [mqtt_connector.py] - mqtt_connector - 277 - Connector "MQTT Broker Connector" subscribe to /ruuvitelemetry/+"
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [tb_device_mqtt.py] - tb_device_mqtt - 141 - connection SUCCESS"
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [mqtt_connector.py] - mqtt_connector - 320 - ()"
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [mqtt_connector.py] - mqtt_connector - 327 - "MQTT Broker Connector" /ruuvitelemetry/+, subscription message id = 1"
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 2 for device *"
```

So I think the regular expression in the mqtt.json "deviceNameTopicExpression":"(?<=telemetry\/)(.*?)" does not fit.

I want to setup everything like in mapping example 2 (https://thingsboard.io/docs/iot-gateway/config/mqtt/), cause the different devices have all the same pretopic "telemetry/" in this case and as subtopic they use their mac addresses. I want to use this mac addresses as the device names in Thingsboard. The addresses contain colons, could that be the reason why the regEx is not working as in the example? I am not not very familiar with RegEx, so please forgive me if the question is a bit stupid
 
**Configuration** (Attach your configuration file)
**Notate: Remove Access token from file if you wanna attach tb_gateway.yaml**

```
  thingsboard:
  host: 127.0.0.1
  port: 1883
  remoteShell: false
  remoteConfiguration: false
  statsSendPeriodInSeconds: 3600
  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: XXXXXXXXXXXX
  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
  - name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json`

**Connector name (If you need help with some connector/converter):**
[e.g. MQTT Connector]
```
{
  "broker": {
    "name":"Default Local Broker",
    "host":"127.0.0.1",
    "port":1884,
    "clientId": "ThingsBoard_gateway",
    "security": {
      "type": "basic",
      "username": "usernameusername",
      "password": "XXXXXXXXXXXXXX"
    }
  },
  "mapping": [
    {
      "topicFilter": "/telemetry/+",
      "converter": {
        "type": "json",
        "deviceNameTopicExpression": "(?<=telemetry\/)(.*?)",
        "deviceTypeTopicExpression": "Device",
        "timeout": 60000,
        "attributes": [],
        "timeseries": [
          {
            "type": "double",
            "key": "gw_mac",
            "value": "${gw_mac}"
          },
          {
            "type": "double",
            "key": "rssi",
            "value": "${rssi}"
          },
          {
            "type": "double",
            "key": "aoa",
            "value": "${aoa}"
          },
          {
            "type": "double",
            "key": "gwts",
            "value": "${gwts}"
          },
          {
            "type": "double",
            "key": "devicets",
```

**Error traceback (If it was raised):**
```
```

**Info traceback (If it was raised):**
```
Sep 22 11:01:36 ecs-application-server-scobytec python3[259278]: ""2021-09-22 11:01:36" - INFO - [tb_gateway_mqtt.py] - tb_gateway_mqtt - 176 - Subscribed to *|* with id 2 for device *"
```

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Thingsboard IoT Gateway version 2.7
 - Python version Python 3.8.10

samson0v, 2021-09-23:
Hi @RudolphAschmoneit, you configured your IoT Gateway right, but if you want to it works, you have to send data to the broker from a device. There are examples with your config in my PC:
[image]

[image]

RudolphAschmoneit, 2021-09-23:
Hi @samson0v I am sending messages from a device to the broker. When I use mosquitto_sub I also see this messages. But thank you for your hint with the DEBUG level in logs, I should have tried this early. I will let you know next week if this could fix my problem. Have a nice weekend.

zbeacon, 2021-10-01:
Hi @RudolphAschmoneit ,

You have a wrong topic filter in your configuration:
`"topicFilter": "/telemetry/+"`
 It looks like you publish to the topic telemetry/+ , but these are different topics, please change topic filter to `"topicFilter": "telemetry/+"` and run the gateway again. Let us know about the results.

RudolphAschmoneit, 2021-10-03:
Hi @zbeacon Thank's for your answer. That helped! And sorry for not seeing this little mistake by myself

closed 2021-10-03 by RudolphAschmoneit