thingsboard/thingsboard-gateway #579: Failed to start thingsboard-gateway.service: Unit thingsboard-gateway.service not found.

santobedi, 2021-09-28:
**Describe the issue**
I've installed the Thingsboard gateway in my NanoPi device. I attempt two times (1) Installation from Pip and (2) Installation from sources (Latest version). Both times, I can see the server attribute of the gateway device in Web UI as active. However, when I run the command   **sudo systemctl start thingsboard-gateway** ,I get **Failed to start thingsboard-gateway.service: Unit thingsboard-gateway.service not found.**. 

How can I solve it? 

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 18.04]
 - Thingsboard IoT Gateway version [e.g. 2.7]
 - Python version[e.g. 3.7]

zbeacon, 2021-09-30:
Hi @santobedi ,

Thank you for your interest i ThingsBoard IoT gateway.
If you installed the gateway from the sources - you don't have a thingsboard-gateway.service file in /etc/system/systemd/ folder, so it is impossible to run the gateway as a daemon. 
In order to solve this, please try to install the gateway using deb package - this installation will add a service file.

santobedi, 2021-10-04:
Hello @zbeacon , 

I installed the gateway using deb package (Raspberry Pi 3, Ubuntu 20.04 LTS) and configured the **tb_gateway.yaml** (Modbus connector) and **modbus.json** files. Then I run s**udo python3 -c 'from thingsboard_gateway.gateway.tb_gateway_service import TBGatewayService; TBGatewayService("./tb_gateway.yaml")'** to save the configuration. I can see the devices been created and latest telemetry in web UI. However, I'm also getting **|ERROR| - [modbus_connector.py] - modbus_connector - __function_to_device - 384 - Exception Response(131, 3, IllegalAddress)"
NoneType: None** and
 **|ERROR| - [sync.py] - sync - connect - 663 - [Errno 2] could not open port /dev/ttyUSB0: [Errno 2] No such file or directory: '/dev/ttyUSB0'"** 

What is happening here? How can I solve it? 

Thank you.

santobedi, 2021-10-04:
In addition, the [Disconnected](https://github.com/thingsboard/thingsboard-gateway/issues/575) issue is still prevailing. Any solutions to it? @samson0v  @zbeacon

zbeacon, 2021-10-13:
Hi @santobedi 

This Exception means that you cannot read this address using function from configuration. Server doesn't allow it. Please try another function or address.

santobedi, 2021-10-13:
Hello @zbeacon ,
Thank you for the clarification. The powertag I use has ASCII as the datatype for many register addresses as given [here](https://www.productinfo.schneider-electric.com/powertaglinkuserguide/powertag-link-user-guide/English/BM_PowerTag%20Link%20D%20User%20Manual_4af62430_T000501355.xml/$/TPC_SynthesisTableIdentificationREF_0000550649). However, thingsboard does not support ASCII as illustrated [here](https://thingsboard.io/docs/iot-gateway/config/modbus/#data-types). For some telemetry/attributes, I used 'string' as datatype instead of ASCII in the modbus.json file. However, it does not work for all the cases. For example, I want to add 'Asset name' (as shown in the attached screenshot) as an attribute of my powertag, using 'string' does not in this case. What can I do in this situation? 
[image]