thingsboard/thingsboard-gateway #584: Thingsboard IoT Gateway -- BACnet Connector [HELP]

moharash, 2021-09-29:
**Describe the issue**
Hi, i am trying to connect Thingsboard gateway to a bacnet ip device. the gateway and the device are on the same switch with same network.

thingsboard iot gateway installed on Ubuntu and all required settings are there.

the IoT gateway already registered to the Thingsboard platform:
[image]

**Configuration** (Attach your configuration file)
[image]

**Connector name (If you need help with some connector/converter):**
BACnet Connector

**Error traceback (If it was raised):**
[image]

**Versions (please complete the following information):**
 - OS: buntu 18.04
 - Thingsboard IoT Gateway version 2.2.4
 - Python version 3.7

moharash, 2021-09-30:
Got below from Connector log

[image]

moharash, 2021-09-30:
service log

[image]

[image]

zbeacon, 2021-09-30:
Hi @moharash ,

Thank you for your interest in ThingsBoard IoT gateway.

It looks like some misconfiguration, could you provide your configuration json file for BACnet connector?

moharash, 2021-09-30:
Hi @zbeacon ,

thank you for your reply,

kindly find attached the bacnet.json file and a 2 screenshots from a BACnet explorer showing the device
[bacnet.txt](https://github.com/thingsboard/thingsboard-gateway/files/7260459/bacnet.txt)
s.
[image]
[image]

moharash, 2021-10-04:
Hi @zbeacon ,

what do u think?

closed 2021-11-15 by zbeacon