thingsboard/thingsboard-gateway #614: modbus type 32float get data error

RenCisco, 2021-11-02:
**Describe the bug**
use modbus tcp get data from device.
configuration:{"tag":"a","type": "32float","functionCode": 4,"objectsCount":2,"address": 0},
{"tag":"b","type": "32float","functionCode": 4,"objectsCount":2,"address": 2}
but the return data  is 	2.4180806300389043E-41. 
the correct data is 229.01.
[image]
[image]

question 1,how can  i solve the above problem?
               2, what does the  "byteOrder": "BIG" in the configuration file  mean.?
               3,what are  the selection values for "byteOrder".

**Connector name (If bug in the some connector):**
[e.g. modbus connector]

```

**Versions (please complete the following information):**
 - OS: [e.g. centos 7]
 - Thingsboard IoT Gateway version [e.g. 2.0]
 - Python version[e.g. 3.7]

zbeacon, 2021-11-02:
Hi @RenCisco ,

Thank you for interest in ThingsBoard IoT gateway. byteOrder parameter gives ability for the gateway to understand how to read value from incoming data bytes. byteOrder has 2 possible values: BIG(Big endian ) and LITTLE(little endian). Usually byteOrder depends on Modbus server configuration. You can change the byteOrder and check the incoming data form the gateway. Also the gateway contains wordOrder parameter with the same parameters. The following table may helpful in order to understand how it works:

Word Order - BIG                     
Byte Order - BIG
`word1 =0x1234 word2 = 0x5678`

Word Order - BIG                      
Byte Order - LITTLE
`word1 =0x3412 word2 = 0x7856`

Word Order - LITTLE                   
Byte Order - BIG

`word1 = 0x5678 word2 = 0x1234`

Word Order - LITTLE                   
Byte Order - LITTLE
`word1 =0x7856 word2 = 0x3412`

RenCisco, 2021-11-03:
Hi @zbeacon  
Thank you so much for your assistance. I can get the data,but i have two new problems.
issue 1. Data gaps occur,with on data lost every other row.
[image]

issue2.The ThingsBoard page did not respond with a blank page when the Modbus configuration file was too many lines  ,
 How many lines can a Modbus configuration file support  ?

### **Here is the configuration file**
[modbusConf.txt](https://github.com/thingsboard/thingsboard-gateway/files/7466533/modbusConf.txt)

closed 2021-11-10 by RenCisco