thingsboard/thingsboard-gateway #618: [HELP] modbus io connect to modbus slave

roypan21, 2021-11-04:
I noticed that i current gateway modbus io only support connect to a modbus server. my device is running as modbus slave. is there any feature in roadmap that support to modbus slave. which also mean make the gateway as modbus server to collect data .
 - Thingsboard IoT Gateway version [e.g. 2.2.4]

zbeacon, 2021-11-12:
Hi @roypan21 ,

Thank you for your interest in ThingsBoard IoT gateway. We are going to implement logic like this, it is added to backlog and will be implemented.

roypan21, 2021-11-17:
@zbeacon Thanks! let me know when it is ready. I would love to help with testing.

samson0v, 2021-12-01:
Hi @roypan21, we updated Gateway Modbus Connector so you can try and test it!
Pay attention, that there is a new config file for Modbus, to use it, you have to run Gateway with the old config file for Modbus and Gateway generate a new config file that will be named as "new_modbus.json" in the `config/` folder. And finally, in the `tb_gateway.yml` config file use the new Modbus config file.

roypan21, 2021-12-03:
@samson0v  I installed IoT Gateway using docker on windows by following the instruction. since the new feature is at master brach. could you provide instructions on how to remove the old version before I [install it from the source](https://thingsboard.io/docs/iot-gateway/install/source-installation/)?

samson0v, 2021-12-03:
Hi @roypan21 , follow the next steps:

1. Clone the master branch
2. Create virtual env using `python3 -m venv venv`
3. Activate virtual env `source venv/bin/activate`
4. Run `pip setup.py install `
5. Configure Gateway
6. Run Gateway

samson0v, 2021-12-17:
Hi @roypan21, did you test the new version of the Modbus Connector, can we close the issue?

roypan21, 2021-12-17:
Hi, 
   I was waiting for my Modbus device come back from field for testing. Have not tested yet . You can resolve it and I will re open a ticket if I see any issue as soon as I start test with my modbus device(expect start in 3 weeks after holiday)
Thanks!

closed 2021-12-17 by zbeacon