thingsboard/thingsboard-gateway #619: [BUG] Opcua connector (subscribe mode) crashes after reading in the list of variables

DejonckheereWard, 2021-11-04:
**Describe the bug**
When using the opcua connector, since versions 2.8 & 2.9, the connector crashes after finishing reading all the variables from the list in opcua.json. When i turn "disableSubscriptions" to true, this doesn't happen.
When installing version 2.7 from scratch, this issue does not appear. But on an install that had 2.8 installed by default, then downgraded to 2.7, the issue also occurs.
I am not sure what is causing the connector to crash.

**Connector name (If bug in the some connector):**
OPC-UA connector

**Error traceback (If available):**
```
Nov 04 09:53:38 RevPi59530 python3[31413]: ""2021-11-04 09:53:38" - |CRITICAL| - [ua_client.py] - ua_client - _receive - 120 - Received an error: MessageAbort(error:StatusCode(BadTcpSecureChannelUnknown), reason:None)"
Nov 04 09:53:38 RevPi59530 python3[31413]: ""2021-11-04 09:53:38" - |ERROR| - [opcua_connector.py] - opcua_connector - run - 155 - Connection failed on connection to OPC-UA server with url 10.10.0.100:4840"
Nov 04 09:53:38 RevPi59530 python3[31413]: ""2021-11-04 09:53:38" - |ERROR| - [opcua_connector.py] - opcua_connector - __search_node - 522 - Request during search has been canceled by the OPC-UA server."
Nov 04 09:53:38 RevPi59530 python3[31413]: ""2021-11-04 09:53:38" - |ERROR| - [opcua_connector.py] - opcua_connector - run - 156 - "The SecureChannelId and/or TokenId are not currently in use."(BadTcpSecureChannelUnknown)"
Nov 04 09:53:38 RevPi59530 python3[31413]: Traceback (most recent call last):
Nov 04 09:53:38 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 131, in run
Nov 04 09:53:38 RevPi59530 python3[31413]:     self.client.connect()
Nov 04 09:53:38 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/client.py", line 274, in connect
Nov 04 09:53:38 RevPi59530 python3[31413]:     self.send_hello()
Nov 04 09:53:38 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/client.py", line 320, in send_hello
Nov 04 09:53:38 RevPi59530 python3[31413]:     raise ack
Nov 04 09:53:38 RevPi59530 python3[31413]: opcua.ua.uaerrors._auto.BadTcpSecureChannelUnknown: "The SecureChannelId and/or TokenId are not currently in use."(BadTcpSecureChannelUnknown)
Nov 04 09:53:39 RevPi59530 python3[31413]: ""2021-11-04 09:53:39" - |ERROR| - [ua_client.py] - ua_client - _call_publish_callback - 537 - Exception while calling user callback: %s"
Nov 04 09:53:39 RevPi59530 python3[31413]: Traceback (most recent call last):
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 535, in _call_publish_callback
Nov 04 09:53:39 RevPi59530 python3[31413]:     callback(response.Parameters)
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/common/subscription.py", line 140, in publish_callback
Nov 04 09:53:39 RevPi59530 python3[31413]:     self.server.publish([ack])
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 482, in publish
Nov 04 09:53:39 RevPi59530 python3[31413]:     self._uasocket.send_request(request, self._call_publish_callback, timeout=0)
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 81, in send_request
Nov 04 09:53:39 RevPi59530 python3[31413]:     future = self._send_request(request, callback, timeout, message_type)
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 72, in _send_request
Nov 04 09:53:39 RevPi59530 python3[31413]:     self._socket.write(msg)
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/common/utils.py", line 118, in write
Nov 04 09:53:39 RevPi59530 python3[31413]:     self.socket.sendall(data)
Nov 04 09:53:39 RevPi59530 python3[31413]: OSError: [Errno 9] Bad file descriptor
Nov 04 09:53:39 RevPi59530 python3[31413]: ""2021-11-04 09:53:39" - |ERROR| - [ua_client.py] - ua_client - _call_publish_callback - 537 - Exception while calling user callback: %s"
Nov 04 09:53:39 RevPi59530 python3[31413]: Traceback (most recent call last):
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 535, in _call_publish_callback
Nov 04 09:53:39 RevPi59530 python3[31413]:     callback(response.Parameters)
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/common/subscription.py", line 140, in publish_callback
Nov 04 09:53:39 RevPi59530 python3[31413]:     self.server.publish([ack])
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 482, in publish
Nov 04 09:53:39 RevPi59530 python3[31413]:     self._uasocket.send_request(request, self._call_publish_callback, timeout=0)
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 81, in send_request
Nov 04 09:53:39 RevPi59530 python3[31413]:     future = self._send_request(request, callback, timeout, message_type)
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 72, in _send_request
Nov 04 09:53:39 RevPi59530 python3[31413]:     self._socket.write(msg)
Nov 04 09:53:39 RevPi59530 python3[31413]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/common/utils.py", line 118, in write
Nov 04 09:53:39 RevPi59530 python3[31413]:     self.socket.sendall(data)
Nov 04 09:53:39 RevPi59530 python3[31413]: OSError: [Errno 9] Bad file descriptor

```

**Versions (please complete the following information):**
 - OS: Raspbian Buster
 - Thingsboard IoT Gateway version: 2.8 & 2.9
 - Python version: 3.7.3

DejonckheereWard, 2021-11-04:
An other traceback that has a different traceback. This is after i uninstalled 2.9 and reinstalled 2.7, while i previously had it working on a fresh 2.7 install.
```
Nov 04 10:36:12 RevPi59530 python3[1620]: Received an error: MessageAbort(error:StatusCode(BadTcpSecureChannelUnknown), reason:None)
Nov 04 10:36:12 RevPi59530 python3[1620]: Received an error: MessageAbort(error:StatusCode(BadTcpSecureChannelUnknown), reason:None)
Nov 04 10:36:12 RevPi59530 python3[1620]: Protocol Error
Nov 04 10:36:12 RevPi59530 python3[1620]: Traceback (most recent call last):
Nov 04 10:36:12 RevPi59530 python3[1620]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 101, in _run
Nov 04 10:36:12 RevPi59530 python3[1620]:     self._receive()
Nov 04 10:36:12 RevPi59530 python3[1620]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 121, in _receive
Nov 04 10:36:12 RevPi59530 python3[1620]:     self._call_callback(0, ua.UaStatusCodeError(msg.Error.value))
Nov 04 10:36:12 RevPi59530 python3[1620]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 131, in _call_callback
Nov 04 10:36:12 RevPi59530 python3[1620]:     .format(request_id, self._callbackmap.keys())
Nov 04 10:36:12 RevPi59530 python3[1620]: opcua.ua.uaerrors._base.UaError: No future object found for request: 0, callbacks in list are dict_keys([1, 2])
Nov 04 10:36:12 RevPi59530 python3[1620]: [STREAM ONLY] 2021-11-04 10:36:12,043 - ERROR - [opcua_connector.py] - opcua_connector - 522 - Request during search has been canceled by the OPC-UA server.
Nov 04 10:36:12 RevPi59530 python3[1620]: exception calling callback for <Future at 0x705b0df0 state=cancelled>
Nov 04 10:36:12 RevPi59530 python3[1620]: Traceback (most recent call last):
Nov 04 10:36:12 RevPi59530 python3[1620]:   File "/usr/lib/python3.7/concurrent/futures/_base.py", line 324, in _invoke_callbacks
Nov 04 10:36:12 RevPi59530 python3[1620]:     callback(self)
Nov 04 10:36:12 RevPi59530 python3[1620]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 201, in clb
Nov 04 10:36:12 RevPi59530 python3[1620]:     response = struct_from_binary(ua.OpenSecureChannelResponse, future.result())
Nov 04 10:36:12 RevPi59530 python3[1620]:   File "/usr/lib/python3.7/concurrent/futures/_base.py", line 423, in result
Nov 04 10:36:12 RevPi59530 python3[1620]:     raise CancelledError()
Nov 04 10:36:12 RevPi59530 python3[1620]: concurrent.futures._base.CancelledError
Nov 04 10:36:22 RevPi59530 python3[1620]: [STREAM ONLY] 2021-11-04 10:36:22,039 - ERROR - [opcua_connector.py] - opcua_connector - 522 - Request during search has been canceled by the OPC-UA server.
Nov 04 10:36:22 RevPi59530 python3[1620]: [STREAM ONLY] 2021-11-04 10:36:22,144 - ERROR - [opcua_connector.py] - opcua_connector - 528 - 'NoneType' object has no attribute 'write'
Nov 04 10:36:22 RevPi59530 python3[1620]: Traceback (most recent call last):
Nov 04 10:36:22 RevPi59530 python3[1620]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 557, in datachange_notification
Nov 04 10:36:22 RevPi59530 python3[1620]:     subscription = self.connector.subscribed[node]
Nov 04 10:36:22 RevPi59530 python3[1620]: KeyError: Node(StringNodeId(ns=2;s=GVL.Fout_Rem_T2))
Nov 04 10:36:22 RevPi59530 python3[1620]: During handling of the above exception, another exception occurred:
Nov 04 10:36:22 RevPi59530 python3[1620]: Traceback (most recent call last):
Nov 04 10:36:22 RevPi59530 python3[1620]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 472, in __search_node
Nov 04 10:36:22 RevPi59530 python3[1620]:     current_node_path = self.get_node_path(current_node)
Nov 04 10:36:22 RevPi59530 python3[1620]:   File "/usr/lib/python3/dist-packages/thingsboard_gateway/connectors/opcua/opcua_connector.py", line 453, in get_node_path
Nov 04 10:36:22 RevPi59530 python3[1620]:     return '\\.'.join(node.get_browse_name().Name for node in node.get_path(200000))
Nov 04 10:36:22 RevPi59530 python3[1620]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/common/node.py", line 415, in get_path
Nov 04 10:36:22 RevPi59530 python3[1620]:     path = self._get_path(max_length)
Nov 04 10:36:22 RevPi59530 python3[1620]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/common/node.py", line 434, in _get_path
Nov 04 10:36:22 RevPi59530 python3[1620]:     refs = node.get_references(refs=ua.ObjectIds.HierarchicalReferences, direction=ua.BrowseDirection.Inverse)
Nov 04 10:36:22 RevPi59530 python3[1620]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/common/node.py", line 369, in get_references
Nov 04 10:36:22 RevPi59530 python3[1620]:     results = self.server.browse(params)
Nov 04 10:36:22 RevPi59530 python3[1620]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 327, in browse
Nov 04 10:36:22 RevPi59530 python3[1620]:     data = self._uasocket.send_request(request)
Nov 04 10:36:22 RevPi59530 python3[1620]:   File "/var/lib/thingsboard_gateway/.local/lib/python3.7/site-packages/opcua/client/ua_client.py", line 81, in send_request
[... cut ...]
Nov 04 10:36:22 RevPi59530 python3[1620]: [STREAM ONLY] 2021-11-04 10:36:22,166 - ERROR - [opcua_connector.py] - opcua_connector - 524 - Broken Pipe. Connection lost.
Nov 04 10:36:22 RevPi59530 python3[1620]: [STREAM ONLY] 2021-11-04 10:36:22,169 - ERROR - [opcua_connector.py] - opcua_connector - 524 - Broken Pipe. Connection lost.
Nov 04 10:36:22 RevPi59530 python3[1620]: [STREAM ONLY] 2021-11-04 10:36:22,172 - ERROR - [opcua_connector.py] - opcua_connector - 524 - Broken Pipe. Connection lost.

```

DejonckheereWard, 2021-11-04:
After more inspection, i'm not sure wether the version of the gateway is the cause of this. I narrowed the bug down to having more than 512 keys in the opcua.json
The moment i have a single extra key, the program crashes. Sorry for confusing it up and thinking it was a bug with a new update.
Is there any way to solve this limit?
The reason I use the subscription is to have quicker updates, when the disableSubscriptions is on True, it takes up to 40-50 seconds per update.

samson0v, 2021-11-09:
Hi @DejonckheereWard, if you want to read all node data, you can try to use .* as a deviceNodePattern.
For example:
[image]
[image]

Thanks for your interest in ThingsBoard IoT Gateway and let us know if it was helping

DejonckheereWard, 2021-11-15:
@samson0v 
No keys have to be specified then, or will all keys have the name "Humidity" in ur example?
I will try this device node pattern this week, i'll update if it works or not (currently other issues i'm working on)

DejonckheereWard, 2021-11-28:
> Hi @DejonckheereWard, if you want to read all node data, you can try to use .* as a deviceNodePattern. For example: [image] [image]
> 
> Thanks for your interest in ThingsBoard IoT Gateway and let us know if it was helping

I did some testing, and i don't seem to be getting the behaviour i wanted using this. What i want is that the keys are automatically detected in the device.
Examples: "Humidity, Temperature"
Then take over the ending path as the keyname.
This is what i tried, but i'm not getting any data from this:
[image]
If i specify the path to a specific variable this does work.

closed 2022-02-03 by zbeacon