thingsboard/thingsboard-gateway #622: Issue with json expression [BUG]

andonoc, 2021-11-04:
**Describe the issue**
I encountered a issue, when i was running the 2.9-1 version.
On the Telemetry tab the keys and values are shown as the JSON expression from the mqtt.json file but they are processed to the right places.
[image]

I reversed now to 2.8-1 and everything works as expected

I checked the Thingsboard-gateway version with this command:
sudo apt install  -s ./python3-thingsboard-gateway.deb

**Configuration** (Attach your configuration file)
```
thingsboard:
  host: 127.0.0.1
  port: 1883
#  remoteShell: false
#  remoteConfiguration: false
#  statsSendPeriodInSeconds: 3600
#  checkConnectorsConfigurationInSeconds: 60
  security:
    accessToken: ************
#  qos: 1
storage:
  type: memory
  read_records_count: 100
  max_records_count: 100000
#  type: file
#  data_folder_path: ./data/
#  max_file_count: 10
#  max_read_records_count: 10
#  max_records_per_file: 10000
connectors:
  - name: MQTT Broker Connector
    type: mqtt
    configuration: mqtt.json

#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus.json
#
#  -
#    name: Modbus Connector
#    type: modbus
#    configuration: modbus_serial.json
#
#  -
#    name: OPC-UA Connector
#    type: opcua
#    configuration: opcua.json
#
#  -
#    name: BLE Connector
#    type: ble
#    configuration: ble.json
#
#  -
#    name: REQUEST Connector
#    type: request
#    configuration: request.json
#
#  -
#    name: CAN Connector
#    type: can
#    configuration: can.json
#
#  -
#    name: BACnet Connector
#    type: bacnet
#    configuration: bacnet.json
#
#  -
#    name: ODBC Connector
#    type: odbc
#    configuration: odbc.json
#
#  -
#    name: REST Connector
#    type: rest
#    configuration: rest.json
#
#  -
#    name: SNMP Connector
#    type: snmp
#    configuration: snmp.json
#  -
#    name: FTP Connector
#    type: ftp
#    configuration: ftp.json
#
#  -
#    name: Custom Serial Connector
#    type: serial
#    configuration: custom_serial.json
#    class: CustomSerialConnector

```

**Connector name :**
```
{
    "broker": {
        "name": "TTN Broker",
        "host": "eu1.cloud.thethings.network",
        "port": 1883,
        "clientId": "ThingsBoard_TTN_gateway",
        "security": {
            "type": "basic",
            "username": "*************",
            "password": "*************"
        }
    },
    "mapping": [
        {
            "topicFilter": "v3/+/devices/+/up",
            "converter": {
                "type": "json",
                "deviceNameJsonExpression": "${end_device_ids.dev_eui}",
                "deviceTypeJsonExpression": "${uplink_message.version_ids.model_id}",
                "timeout": 60000,
                "attributes": [
                    {
                        "type": "string",
                        "key": "brand",
                        "value": "${uplink_message.version_ids.brand_id}"
                    }

                ],
                "timeseries": [
                    {
                        "type": "double",
                        "key": "Battery",
                        "value": "${uplink_message.decoded_payload.Bat}"
                    },
                    {
                        "type": "double",
                        "key": "SoilWater",
                        "value": "${uplink_message.decoded_payload.water_SOIL}"
                    },
                    {
                        "type": "double",
                        "key": "SoilTemperature",
                        "value": "${uplink_message.decoded_payload.temp_SOIL}"
                    }
                ]
            }
        }
    ]

}

```

**Versions (please complete the following information):**
 - OS: Ubuntu 20.04.3 LTS
 - Thingsboard IoT Gateway version 2.9-1
 - Python version 3.8.10

AngeLinuX99, 2021-11-04:
Yes, same problem to me. 

Some values are displayed correctly, but others are not.

[image]

closed 2021-11-05 by zbeacon

rduivenvoorde, 2021-11-22:
Ah, thanks for fixing, I'm also bitten by this one...