thingsboard/thingsboard-gateway #629: Does tb_Gateway support IEC104?

RenCisco, 2021-11-12:
issue:1.how many lines are supported by single Modbus configuration file ? Instead of configuring multiple Modbus files
         2.Does Gateway support IEC104?

**Versions (please complete the following information):**
 - OS: [e.g. centos7]
 - Thingsboard IoT Gateway version [e.g. 2.2.4]
 - Python version[e.g. 3.7]

samson0v, 2021-11-12:
Hi @RenCisco,
1. What logs does Gateway output? 
2. Gateway doesn't support the IEC 104 protocol.

RenCisco, 2021-11-15:
@samson0v

gateway logs are "2021-11-12 16:35:31" - ERROR - [modbus_connector.py] - modbus_connector - 185 - Connection lost! Reconnecting..."

issue:1.How do I obtain hexadecimal Modbus packets?

samson0v, 2021-11-15:
Hi @RenCisco, you can't send bytes to TB, we are using UTF-8 format, but you can parse the data in RuleChain.

closed 2021-11-25 by zbeacon