thingsboard/thingsboard-gateway #630: [HELP] Not using 'on_attributes_update' from custom SNMP Connector

konradSC, 2021-11-12:
**Describe the issue**
I am using a custom SNMP connector which is working fine when polling devices and sending telemetry to TB. The issue is when we update a shared attribute in TB. The tb-gateway tries to use the "on_attributes_update" method from the default snmp connector instead of our custom connector. 

This is happening with the latest docker image. 

**Configuration** (Attach your configuration file)
**Notate: See attached

**Connector name (If you need help with some connector/converter):**
Custom SNMP

**Error traceback (If it was raised):**
```
'deviceName'
Traceback (most recent call last):
  File "<input>", line 2, in <module>
KeyError: 'deviceName'
```

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version [2.2.4] Docker
 - Python version[e.g. 3.7]
[CustomSNMPConnector.txt](https://github.com/thingsboard/thingsboard-gateway/files/7528622/CustomSNMPConnector.txt)

samson0v, 2021-11-15:
Hi @konradSC, are you sure that the custom connector is loaded?
Can you, please, send gateway logs on debug level (if you want to see debug logs you can set it in the config/logs.conf file)?

samson0v, 2021-11-15:
If you are using docker you can try to use this [solution](https://github.com/thingsboard/thingsboard-gateway/issues/615) for loading custom connector 

This bug was fixed but gateway container not updated yet.

samson0v, 2021-11-29:
Hi @konradSC, did it help, can we close the issue?

konradSC, 2021-11-29:
Hi @samson0v. The custom connector loads without issue and works properly to poll the device. It seems to be using the wrong snmp connector on attribute updates.

closed 2021-11-30 by zbeacon