thingsboard/thingsboard-gateway #631: [HELP] Add new RPC method

gohm-tech, 2021-11-15:
Hi,
We have different RPC methods from the RPC methods found on the thingsboard-gateway. How can we add new RPC methods?
We followed the steps below to add a new method, but we were unsuccessful.

1. We stopped the service.
2. We have made changes to the attached file below.
[tb_gateway_service.txt](https://github.com/thingsboard/thingsboard-gateway/files/7537749/tb_gateway_service.txt)

3. We started the service.
After these steps, we observed that the service was not working.

Best Regards,
Gohm Team

** Server Environment**

own setup

docker deployment

OS: Linux(Ubuntu 20.04)
** Client Environment**

OS: Linux(Ubuntu)
Browser: Chrome, Firefox
Version Ubuntu Version 20.04

samson0v, 2021-11-15:
Hi @gohm-tech, yes, you can do it in that way. In the future, we will add a default tool for adding RPCs.
Please, send the gateway logs and errors that you took.
Thanks for your interest in ThingsBoard IoT Gateway.

gohm-tech, 2021-11-18:
Hi, @samson0v 
Thanks for your helping.
Best Regards,
**Gohm Team**

closed 2021-11-18 by gohm-tech