thingsboard/thingsboard-gateway #634: [BUG] Timeout fields added in server side rpc do not reach thingsboard-gateway

gohm-tech, 2021-11-18:
Hi,
I need to get the timeout field when two-way RPC is received to thingsboard-gateway.

In the screenshot below, there is a two-way RPC request that I sent with the help of a swagger-ui.
[image]

I have a screenshot of listening to the RPC request with HTTP Get.
[image]

There is a screenshot showing RPC request logs.
[image]

**Configuration**
[tb_gateway_service.txt](https://github.com/thingsboard/thingsboard-gateway/files/7562035/tb_gateway_service.txt)

**Connector name**
MQTT Connector

**Versions:**
OS: Raspbian GNU/Linux 10 (buster)
Thingsboard IoT Gateway version [e.g. 2.7]
Python version 3.7.3

Best Regards,
**Gohm Team**

zbeacon, 2021-11-18:
Hi @gohm-tech ,

Thank you for your interest in ThingsBoard IoT gateway.
Unfortunately, it is not possible to get this value from incoming request on the gateway, because timeout field and persistent flag are not send to devices from ThingsBoard. You can just put timeout as a parameter into params and use them on the devices side.

gohm-tech, 2021-11-18:
Hi @zbeacon 
Thanks for your quick reply,

When I examined the code, I noticed the lines in the picture below. 
[image]

Here, the "timeout" variable in the content variable is added to the arguments variable. This arguments variable contains the "params" field in the RPC request body we sent. This arguments variable is added as an argument to the RPC method function. If I was going to add the timeout value into the "params" variable, I think there was no need to write such an if state. Could this be a bug on the server-side? What is the reason for adding this if-statement?

Best Regards,
**Gohm Team**

zbeacon, 2021-11-18:
Hmm, let me check this point.

zbeacon, 2021-11-18:
Unfortunately there is no timeout field in RPC request message, this if was redundant.

gohm-tech, 2021-11-18:
Hi,

@zbeacon Thank you for your interest.

Best Regards,
**Gohm Team**

closed 2021-11-18 by gohm-tech