thingsboard/thingsboard-gateway #635: [BUG] OPC-UA connector ERROR. 'uacrypto' not defined when running docker image.

artturimW, 2021-11-19:
**Describe the bug**
Trying to connect to PLC with OPC-UA connector. Using docker-compose with thingsboard/tb-gateway:latest image. 
Trying to connect using following config:  

security: Basic256Sha256
type: cert.PEM
cert: aaa
publickey: bbb
mode: SignAndEncrypt

Same config is working when running tb_gateway without docker. I also managed to get tb_gateway working with docker when using only "anonymous" security.

Error says  'uacrypto' is not defined and then ua_client returning "error:StatusCode(BadSecurityPolicyRejected)

Docker-compose:
```
version: '3.8'

services:
  tbg-1:
    image: thingsboard/tb-gateway:latest
    restart: always
    networks:
      - gateway
    volumes:
      - ./config:/thingsboard_gateway/config
      - ./extensions:/thingsboard_gateway/extensions
      - tbg-1-logs:/thingsboard_gateway/logs:z

networks:
  gateway:
    driver: bridge

volumes:
  tbg-1-logs:
    driver: local
```

**Connector name (If bug in the some connector):**
OPC-UA Connector

**Error traceback (If available):**
[image]

**Versions (please complete the following information):**
 - OS: [e.g. Ubuntu 20.04]
 - Thingsboard IoT Gateway version 2.9 /latest
 - Python version 3.7

samson0v, 2021-11-23:
Hi @artturimW, it seems that docker can't find the .pem file, make sure that you bind your certificate to docker volume.
Thanks for your interest in ThingsBoad IoT Gateway.

samson0v, 2021-11-29:
Hi @artturimW, did it help, can we close the issue?

artturimW, 2021-11-30:
Could not get it work, tried to conf pem files with many different ways. And checked that the .pem file actually exists in the container. Made my own image for gateway and now it is working.

samson0v, 2021-12-07:
Hi @artturimW, what did you change to get it to work?

closed 2022-05-20 by imbeacon